dotnet-bluetooth-le/dotnet-bluetooth-le #612: What is the actual type of Device, Service, Characteristic etc?

gfmoore, 2022-07-06:
I'm getting somewhere with this code on .net Maui, but as a newbie I'm struggling to organise my design.

You often use var, but I would prefer to use a more global type, but I simply cannot find a way to declare the variable.

So instead of 
```
{
   var deviceHRM = = await adapter.ConnectToKnownDeviceAsync(guid, cp, cts.Token); 
}
```

I would prefer to write
```
sometype deviceHRM

{
  deviceHRM = = await adapter.ConnectToKnownDeviceAsync(guid, cp, cts.Token); 
}
```

Please put me out of my misery ;)

Same with service and characteristic. I am connecting to 4 or 5 devices!!!

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

gfmoore, 2022-07-06:
Is it
```
IDevice deviceHRM;
```
?

MattVon, 2022-08-16:
[image]

The return type is IDevice.

closed 2022-08-16 by gfmoore