dotnet-bluetooth-le/dotnet-bluetooth-le #615: ServiceBase.GetCharacteristicsAsync() throws ArgumentException: length

TomasMalcanek, 2022-08-18:
This error is happening extremely often and only on Android.

It's easy workaround, just adding a try catch and calling the method again fixes it.

## Steps to reproduce

1. Request characteristics

## Expected behavior

Characteristics should be loaded.

## Actual behavior

Method throws an exception.

### Crashlog

**ServiceBase.GetCharacteristicsAsync ()
System.ArgumentException: length**

```
System 
Array.Copy (System.Array sourceArray, System.Int32 sourceIndex, System.Array destinationArray, System.Int32 destinationIndex, System.Int32 length)

System.Collections.Generic 
List`1[T].CopyTo (T[] array, System.Int32 arrayIndex)

System.Collections.Generic.List`1[T] 
System.Collections.Generic.List`1[T]..ctor (System.Collections.Generic.IEnumerable`1[T] collection) [0x0003b] in <ca00b961ae9e43f0bc6166bec1195ff4>:0

System.Linq 
Enumerable.ToList[TSource] (System.Collections.Generic.IEnumerable`1[T] source)

Plugin.BLE.Abstractions 
ServiceBase.GetCharacteristicsAsync ()

Plugin.BLE.Abstractions 
ServiceBase.GetCharacteristicAsync (System.Guid id)
```

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:**

- Android 10
- Android 11
- Android 12
- Android 13

**Device:**

- Galaxy A52s 5G
- Galaxy S21 5G
- P30 Pro
- Galaxy A51
- HUAWEI P30 lite
- Galaxy S9
- Galaxy S10
- Galaxy S20 FE
- Galaxy S20 FE 5G
- Galaxy A50

TomasMalcanek, 2022-08-22:
I found a better error message today that says more than just "length". The stacktrace and rest of the exception is the same.

ServiceBase.GetCharacteristicsAsync ()
System.ArgumentException: Destination array was not long enough. Check destIndex and length, and the array's lower bounds Parameter name: destinationArray

janusw, 2022-08-28:
Could you provide a reproducer, please?

TomasMalcanek, 2023-06-05:
I don't think it's a duplicate, it might be worth a mention, but the stack traces and when it occurs are different.

No clear reproducer made yet. Collecting information about this is slow as it's again one of those bugs that don't happen on the dev side.