dotnet-bluetooth-le/dotnet-bluetooth-le #617: PluginBLE 2.1.3 breaks Android app; close() missing?

petemillermultimedia, 2022-08-25:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Hmm, my app works with plugin version 2.1.2, breaks with 2.1.3. Reproduce by installing 2.1.3

2.

3.

## Expected behavior
App repeatedly connects to devices to capture data over separate sessions.
Debug log of my app shows "[Bluetooth] close()" near the end of the session.

## Actual behavior
App connects to device successfully once; subsequent connections hang.
Debug log of my app no longer shows "[Bluetooth] close()" near the end of the session.

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin: 2.1.3

**Platform:** Android 9 and Android 12

**Device:** Samsung G8, Pixel 5

Observation: I no longer see the close() operation at the end of a device session. I notice that a new bit of conditional code was added around the close() call in Plugin.BLE.Android.GattCallback.cs

                    if (!_device.ConnectParameters.AutoConnect)
                    {
                        CloseGattInstances(gatt);
                    }

The AutoConnect parameter is set to true for my device, so CloseGattInstances() isn't called. I'm not sure how this is supposed to work, but I can't find any other differences to account for the failure of 2.1.3 when 2.1.2 works, so I suspect that the missing close() call leads to my problem. Any insight?

petemillermultimedia, 2022-08-25:
After doing some digging, I suspect that the behavior described above is intentional, i.e., if AutoConnect is true there shouldn't be a close(). If so, then is it also the case that the DeviceDiscovered callback is only called once for a device, whereas DeviceConnected may occur repeatedly? Sorry to be getting Bluetooth 101 from you, but if these assumptions are true then I understand where my problem is.

janusw, 2022-08-28:
> After doing some digging, I suspect that the behavior described above is intentional, i.e., if AutoConnect is true there shouldn't be a close().

Yes, the change in behavior is probably due to PR #519.  Upon disconnection, the GATT is only closed if `AutoConnect` is `false`. I assume it is `true` in your case?

> If so, then is it also the case that the DeviceDiscovered callback is only called once for a device, whereas DeviceConnected may occur repeatedly?

I think so.

> Sorry to be getting Bluetooth 101 from you, but if these assumptions are true then I understand where my problem is.

Alright, so we can close this issue?

petemillermultimedia, 2022-08-30:
I'm using AutoConnect incorrectly for the use case. Closing this issue.

closed 2022-08-30 by petemillermultimedia