dotnet-bluetooth-le/dotnet-bluetooth-le #619: Adapter.IsScanning is always true if exception occurs

andrey-val-rodin, 2022-09-03:
In UWP, StartScanningForDevicesAsync throws system exception if Bluettoth is off. After that, Adapter.IsScanning is always true and there is no way to stop scanning.

## Steps to reproduce

1. Turn Bluetooth off
2. Call StartScanningForDevicesAsync in UWP app

## Expected behavior
Adapter.IsScanning is false

## Actual behavior
Adapter.IsScanning is true

### Crashlog

System.Exception
  HResult=0x800710DF
  Message = The device is not ready for use. (Exception from HRESULT: 0x800710DF)
  Source = Windows
  Stack:
   in Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisementWatcher.Start()
   in Plugin.BLE.UWP.Adapter.StartScanningForDevicesNativeAsync(ScanFilterOptions scanFilterOptions, Boolean allowDuplicatesKey, CancellationToken scanCancellationToken)
   in Plugin.BLE.Abstractions.AdapterBase.<StartScanningForDevicesAsync>d__49.MoveNext()
   in RotatingTable.Xamarin.Services.BluetoothService.<ScanAsync>d__62.MoveNext()
## Configuration

**Version of the Plugin:** 2.2.0 Pre5

**Platform:** UWP

**Device:** desktop computer with Windows 11 (build 22000)

JacobEemagine, 2026-09-09:
I just wanted to bump this issue, since it is rather disruptive. If Bluetooth is off, the Exception is not a TimeoutException, so IsScanning will always be true, which prevents future scans from going out:

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b08c8ce584e529f857922a3c086d03474da8e0c9/Source/Plugin.BLE/Shared/AdapterBase.cs#L155

andrey-val-rodin, 2026-10-14:
I verified this in MAUI test application on .NET 9.0.
If Bluetooth is turned off, the StartScanningForDevicesAsync method throws a system exception.
After that, the Adapter.IsScanning property is false.
**This is expected behavior.**

> Exception details:
> System.Runtime.InteropServices.COMException
>   HResult=0x800710DF
>   Сообщение = 
>   Источник = WinRT.Runtime
>   Трассировка стека:
>    в WinRT.ExceptionHelpers.<ThrowExceptionForHR>g__Throw|38_0(Int32 hr)
>    в WinRT.ExceptionHelpers.ThrowExceptionForHR(Int32 hr)
>    в ABI.Windows.Devices.Bluetooth.Advertisement.IBluetoothLEAdvertisementWatcherMethods.Start(IObjectReference _obj)
>    в Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisementWatcher.Start()
>    в Plugin.BLE.Windows.Adapter.StartScanningForDevicesNativeAsync(ScanFilterOptions scanFilterOptions, Boolean allowDuplicatesKey, CancellationToken scanCancellationToken)
>    в Plugin.BLE.Abstractions.AdapterBase.<StartScanningForDevicesAsync>d__55.MoveNext()
>    в PluginTest.MainPage.<OnCounterClicked>d__3.MoveNext() в D:\Source\PluginTest\PluginTest\MainPage.xaml.cs:строка 19
> 
>   Изначально это исключение было создано в этом стеке вызовов: 
>     WinRT.ExceptionHelpers.ThrowExceptionForHR.__Throw|38_0(int)
>     WinRT.ExceptionHelpers.ThrowExceptionForHR(int)
>     ABI.Windows.Devices.Bluetooth.Advertisement.IBluetoothLEAdvertisementWatcherMethods.Start(WinRT.IObjectReference)
>     Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisementWatcher.Start()
>     Plugin.BLE.Windows.Adapter.StartScanningForDevicesNativeAsync(Plugin.BLE.Abstractions.ScanFilterOptions, bool, System.Threading.CancellationToken)
>     Plugin.BLE.Abstractions.AdapterBase.StartScanningForDevicesAsync(Plugin.BLE.Abstractions.ScanFilterOptions, System.Func<Plugin.BLE.Abstractions.Contracts.IDevice, bool>, bool, System.Threading.CancellationToken)
>     PluginTest.MainPage.OnCounterClicked(object, System.EventArgs) в MainPage.xaml.cs

janusw, 2026-10-14:
>         I verified this in MAUI test application on .NET 9.0.
>         If Bluetooth is turned off, the StartScanningForDevicesAsync method throws a system exception.
>         After that, the Adapter.IsScanning property is false.
>         **This is expected behavior.**

@andrey-val-rodin That is with PR #1007, I assume? If yes: Thanks for testing, I'll merge it soon.

andrey-val-rodin, 2026-10-14:
Yes, PR https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/1007

closed 2026-10-14 by janusw

janusw, 2026-11-10:
Fixed with release [3.3.0-beta.3](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.3.0-beta.3).