dotnet-bluetooth-le/dotnet-bluetooth-le #620: Impossible to reconnect device in UWP

andrey-val-rodin, 2022-09-05:
If you connect device and then disconnect it, trying to connect the device again will result in unhandled exception.

## Steps to reproduce

1. Run sample app
2. Connect some device->device connected->Ok
3. Click back arrow in top left corner
4. Click on device again and press Disconnect
5. Try to connect the device ->UnhandledException event fired

## Expected behavior
Second connection is successful

## Actual behavior
Unhandled exception

### Crashlog

-		Exception	{"The object has been closed. (Exception from HRESULT: 0x80000013)"}	System.Exception {System.ObjectDisposedException}
+		Data	{System.Collections.ListDictionaryInternal}	System.Collections.IDictionary {System.Collections.ListDictionaryInternal}
		HResult	-2147483629	int
		HelpLink	null	string
+		InnerException	null	System.Exception
		Message	"The object has been closed. (Exception from HRESULT: 0x80000013)"	string
		ObjectName	""	string
		Source	null	string
		StackTrace	null	string
		TargetSite	null	System.Reflection.MethodBase

## Configuration

**Version of the Plugin:** 2.2.0 Pre5

**Platform:** UWP

**Device:** desktop computer with Windows 11 (build 22000)

janusw, 2022-12-10:
Could this be related to the disconnection issues of #423, #528 and #536? Or is it a separate problem? Does the disconnection operation actually succeed in your case?

Does this still occur with 3.0.0-beta.2?

closed 2023-08-05 by janusw

janusw, 2023-08-05:
Fixed via #721 in [3.0.0-beta.5](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.5).

commit e984ecc1 (2023-08-05): Don't use device.Dispose() on windows in DisconnectDeviceAsync to free resources. Fixes #620