dotnet-bluetooth-le/dotnet-bluetooth-le #621: Xamarin Android App - Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED

ManikandanK-AMI, 2022-09-07:
I Need to develop an app to communicate with a sensor-based hardware component.

I've tried the POC project using the Plugin.Ble and Implemented basic project requirements and Tested the Application in both android and iOS

In the iOS App, can able to send and receive data. But the Android app throws the exception while sending the data.

As per the document, I've sent the data on the MainThread. But it throws the same exception.

## Steps to reproduce

1. Scan for available BT devices
2. Connect to the device
3. GetServiceAsync for a specific characteristic
4 . Assign event handler for ValueUpdated event for that characteristic
5. If characteristic CanUpdate is true, StartUpdatesAsync
6. Attempt to write data to that characteristic using WriteAsync

## Expected behavior
Data is successfully written to the characteristic. In this case, this is the method of transparent data communication between the app and the BT device.

## Actual behavior
can't able to send data to android devices. But it is working fine in iOS App

### Crashlog

Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.
  at Plugin.BLE.Android.Characteristic.InternalWrite (System.Byte[] data) [0x0004a] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:110 
  at Plugin.BLE.Android.Characteristic+<>c__DisplayClass19_0.<WriteNativeAsync>b__0 () [0x00000] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:81 
  at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token) [0x000b3] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\Utils\TaskBuilder.cs:48 
  at Plugin.BLE.Android.Characteristic.WriteNativeAsync (System.Byte[] data, Plugin.BLE.Abstractions.CharacteristicWriteType writeType) [0x000b2] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:80 
  at Plugin.BLE.Abstractions.CharacteristicBase.WriteAsync (System.Byte[] data, System.Threading.CancellationToken cancellationToken) [0x00061] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\CharacteristicBase.cs:92 
  at XamBlePluginPOC.BluetoothCommunicationPage.SendCommandButton_Clicked (System.Object sender, System.EventArgs e) [0x00174] in D:\Xamarin_Sample_Projects\XamBlePluginPOC\BluetoothCommunicationPage.xaml.cs:133
  

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** Android ( Tested in 9 ,10 ,11 and 12)

**Device:** Redmi Note 5 Pro, Samsung Galaxy M52

ManikandanK-AMI, 2022-09-07:
Source Code :

`using Plugin.BLE.Abstractions.Contracts;
using System;
using System.Diagnostics;
using System.Text;
using System.Threading;
using System.Threading.Tasks;
using Xamarin.Forms;
using Xamarin.Forms.Xaml;
using XamarinEssentials = Xamarin.Essentials;

namespace XamBlePluginPOC
{
    [XamlCompilation(XamlCompilationOptions.Compile)]
    public partial class BluetoothDataPage : ContentPage
    {
        private IDevice _ConnectedDevice;

        private IService _Service;
		
		private ICharacteristic sendCharacteristic;
        private ICharacteristic receiveCharacteristic;

        public BluetoothDataPage(IDevice connectedDevice)
        {
            InitializeComponent();
            _ConnectedDevice = connectedDevice;
            InitButton.IsEnabled = !(ScanButton.IsEnabled = false);
        }

      

        private async void InitalizeCommandButton_Clicked(object sender, EventArgs e)
        {
            try
            {

               
                _Service = await _ConnectedDevice.GetServiceAsync(AppDefault.ServiceId);             

                if (_Service != null)
                {

                   
                   sendCharacteristic = await _Service.GetCharacteristicAsync(AppDefault.SendId);

                    receiveCharacteristic = await _Service.GetCharacteristicAsync(AppDefault.ReceiveId);
                    if (receiveCharacteristic != null)
                    {
                        var descriptors = await receiveCharacteristic.GetDescriptorsAsync();

                        XamarinEssentials.MainThread.BeginInvokeOnMainThread(async () =>
                        {
                            receiveCharacteristic.ValueUpdated += async (o, args) =>
                            {
                                var receivedBytes = args.Characteristic.Value;

                                Output.Text += Encoding.UTF8.GetString(receivedBytes, 0, receivedBytes.Length) + Environment.NewLine;
                            };
                        });

                        if (XamarinEssentials.MainThread.IsMainThread)
                        {
                            await receiveCharacteristic.StartUpdatesAsync();
                        }
                        else
                        {
                            XamarinEssentials.MainThread.BeginInvokeOnMainThread(async () =>
                            {
                                await receiveCharacteristic.StartUpdatesAsync();
                            });
                        }
                        InitButton.IsEnabled = !(ScanButton.IsEnabled = true);
                       
                    }
                }
                else
                {
                    				
					await DisplayAlert("Alert","Service not Available","Okay");
                }
            }
            catch (Exception ex)
            {
                Debug.WriteLine($"{ex}");
            }
        }

        
        private async void SendCommandButton_Clicked(object sender, EventArgs e)
        {
            try
            {

                string protocol = "$0F1066 0x20";
                var bytes = Encoding.ASCII.GetBytes($"{protocol}\r\n");
                
                if (sendCharacteristic != null)
                {
                  

                    if (XamarinEssentials.MainThread.IsMainThread)
                    {
                        
                        var result = await sendCharacteristic.WriteAsync(bytes);
                    }
                    else
                    {
                        XamarinEssentials.MainThread.BeginInvokeOnMainThread(async () =>
                        {
                            var result = await sendCharacteristic.WriteAsync(bytes);
                        });
                    }
                }
            }
            catch (Exception ex)
            {
                Debug.WriteLine($"{ex}");

            }

            
        }

        
    }
}`

TomasMalcanek, 2022-11-29:
Hello, I think those errors mostly occur when read and write operations are being made on the same time.
I reworked my client so each read and write locks everything, so every other read/write has to wait for the previous one to finish.

If something happens currently it's only Gatt error 133, and this kind of errors are gone.
(ps I locked pretty much every operation with this plugin to be sure)