dotnet-bluetooth-le/dotnet-bluetooth-le #622: IOS Plugin Issue

SanjayP1234007, 2022-09-09:
Hi Sven-Michael,

I am using this plugin in Xamarin IOS but i found that when i try to use this "await _adapter.StartScanningForDevicesAsync();" ,application gets closed,without throwing any error.I would like to know,can i use this plugin for IOS.I am using Iphone SE(Version 15+).I am using visual studio 2022.
I am using this sample. please find the sample link:

https://drive.google.com/file/d/1Oo4v3VvzXCyT3nwzfaAmGuk1vjBEo_FV/view?usp=sharing

if you have any solution, please provide me.or if this plugin does not have support for Ios then please let me know.

Also please confirm.could this plugin be used for .NET MAUI, as i am migrating my Xamarin code to MAUI.but when i do this i am getting the following message:

"This functionality is not implemented in the portable version of this assembly. You should reference the NuGet package from your main application project in order to reference the platform-specific implementation."

TimoKinnunen, 2022-09-19:
Try this and adopt code from it!

MauiXAMLBluetoothLE

.Net MAUI app using Plugin.BLE for Polar H10 Heart Rate Sensor. Bluetooth LE working solution on Android. I must ask James Montemagno for help to get permissions right on iOS and Windows.

To read code and understand it is simple.

This app might bring closer Bluetooth LE to .Net MAUI. I hope so.

https://github.com/TimoKinnunen/MauiXAMLBluetoothLE

janusw, 2022-10-27:
> Can this plugin be used for .NET MAUI, as i am migrating my Xamarin code to MAUI.but when i do this i am getting the following message:

Not yet, but there is PR #614, which updates it to .NET 6 (thus making it compatible for use with MAUI). Hope it will be merged soon.

janusw, 2022-11-15:
It seems that #614 did actually not fix this problem, but #638 should.

@SanjayP1234007, can you confirm that your problem is gone on current master? (using the nupkg from here: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/3470893880#artifacts)

janusw, 2022-11-17:
Since we have sufficient evidence by now that this kind of problem is fixed via #638 (both on iOS and Windows), I will close this issue. In case the problem persists, please open a new issue (with more details preferably).

closed 2022-11-17 by janusw