dotnet-bluetooth-le/dotnet-bluetooth-le #626: internal static Exception NotImplementedInReferenceAssembly() { return new NotImplementedException("This functionality is not implemented in the portable version of this assembly. You should reference the NuGet package from your main application project in order to reference the platform-specific implementation."); }

TimoKinnunen, 2022-09-26:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

TimoKinnunen, 2022-09-26:
What to do to get Plugin.BLE to work on Windows?

I have a working .Net MAUI sample on Android but Windows is difficult?
https://github.com/TimoKinnunen/MauiXAMLBluetoothLE

naryk86, 2022-09-29:
I am also interested in this topic.
I think that is something that @janusw is looking at with jamesmontemagno... I see that there is a specific pull request ( [LINK](https://github.com/xabre/xamarin-bluetooth-le/pull/614))

TimoKinnunen, 2022-09-29:
Janus Weil and James Montemagno are guys with dynamite in their fingertips.

janusw, 2022-10-27:
> What to do to get Plugin.BLE to work on Windows?

Version 2.2.0-pre5 has some beta support for Xamarin.UWP, but 2.1.3 has not. It is not fully finished, but most basic things should work. If you find something that does not work, please open an issue and give us the details.

> I have a working .Net MAUI sample on Android but Windows is difficult? https://github.com/TimoKinnunen/MauiXAMLBluetoothLE

MAUI does not work yet, but should hopefully soon, see #614.

janusw, 2022-11-15:
This problem should be fixed with #638.

@TimoKinnunen, can you confirm that your problem is gone on current master? (using the nupkg from here: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/3470893880#artifacts)