dotnet-bluetooth-le/dotnet-bluetooth-le #628: Unable to receive complete data from the Arduino BLE HM-10 module to Xamarin app

rauleteee, 2022-10-05:
Hi, I am trying to send a string from xamarin app(c#) to arduino BLE HM-10module, as follows: 
```
public async Task sendString()
        {
            //Guid characteristics = 	Id	{0000ffe1-0000-1000-8000-00805f9b34fb}	System.Guid
            //Guid services = 	Id	{0000ffe0-0000-1000-8000-00805f9b34fb}	System.Guid

            var stringToSend = "Hello World! :)";

            var services = await deviceConn.GetServicesAsync();

            service = await deviceConn.GetServiceAsync(Guid.Parse("0000ffe0-0000-1000-8000-00805f9b34fb"));
            
            var characteristics = await service.GetCharacteristicsAsync();
            characteristic = await service.GetCharacteristicAsync(Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb"));
           
            try
            {
                

                //Write message
                byte[] bytes = Encoding.ASCII.GetBytes(stringToSend).Take(20).ToArray();
                
               
                if (MainThread.IsMainThread)
                {

                    var result = await characteristic.WriteAsync(bytes);
                }
                
                else
                {
                    MainThread.BeginInvokeOnMainThread(async () =>
                    {
                       var result = await characteristic.WriteAsync(bytes);
                    });
                }

                

                msgSended = "Message sended:" + stringToSend;

                //Receive message

                String mesRec = "";

                MainThread.BeginInvokeOnMainThread(async () =>
                {
                    characteristic.ValueUpdated += async (o, args) =>
                    {
                        var receivedBytes = args.Characteristic.Value;

                        mesRec += Encoding.ASCII.GetString(receivedBytes, 0, receivedBytes.Length);
                    };
                });

                if (MainThread.IsMainThread)
                {
                    await characteristic.StartUpdatesAsync();
                }
               
                msgReceived = "Msg received: " + mesRec;

            }
            catch (Exception ex)
            {
                Console.WriteLine(ex);
            }
            
        }
```
At the moment I press the button of sending the string, it is received by BLE in arduino (kind of tx-rx bridge). My goal is to resend this received string to the app on xamarin.
The following code is arduino's behaviour: 
```
#include <SoftwareSerial.h>

SoftwareSerial BT(6,7); //RX, TX

void setup(){
  int baudiosArduino = 9600;
  // Serial USB
  Serial.begin(baudiosArduino);
  while (!Serial) {
    }// espera conexión
  Serial.println("\n USB <->PC, Listo..!!");
  
  //Puerto Serial Módulo Bluetooth
  int baudiosBtLE = 9600;
  BT.begin(baudiosBtLE);
  BT.println("Listo Bluetooth a Tablet/móvil");
  
}

void loop(){
  
  
  if (BT.available() > 0 ){

      delay(10);
     char c = BT.read();
      BT.write(c);
    
      
    }
  
  
}
```
I found two problems:
1- I only receive a portion of the string, and it changes its length when delay is changed (I have already tried to change several times the delay, and it only get worse answer from ble :) ) :
```
Sended string: Hello World! :)
Received string: Hello Wor
```
2- If I press twice the button, nothing is returned to de xamarin app, but if I disconnect, and then connect another time,I can send/receive the semi-string one more time:
```
Sended string: Hello World! :)
Received string: 
```

Anybody knows what is the issue? I will be attentive to your answers.

Thanks in advance,
Raúl.

## Expected behavior
```
Sended string: Hello World! :)
Received string: Hello World! :)
```

## Actual behavior
 > 1st time pressing the button:
```
Sended string: Hello World! :)
Received string: Hello Wor
```
 > 2nd time pressing the button:
```
Sended string: Hello World! :)
Received string: 
```
### Crashlog

Nothing
## Configuration

**Version of the Plugin:** 2.1.3 (Nuget Package)

**Platform:** iOs 16.0

**Device:** iPhone 12 Pro Max

rauleteee, 2022-10-12:
I forgot to mention that I finally reach a possible solution for strings that have less than 20 B ( I suppose it is the bluetooth buffer by trial and error), which was using the function .readStringUntil(character), which specifies to the BLE module when it should stop listening to the string, and it works perfectly fine, only when de string has less or equal  than 20 characters:
```
String c;
void loop(){
  if (BT.available() > 0 ){
    c = BT.readStringUntil('\n');
     BT.print(c);
  }
  }
```
[image]
Here is a snap of my demo which shows how I finally send the string and receive it perfectly. At the time the string > 20B, I will only receive rest of the bytes that appear after the first 20 chars.

This is the final send/receive function in xamarin forms:
```
/******************************
         * BLUETOOTH DATA FLOW
         ******************************/
        public async Task sendString()
        {
            //Guid characteristics = 	Id	{0000ffe1-0000-1000-8000-00805f9b34fb}	System.Guid
            //Guid services = 	Id	{0000ffe0-0000-1000-8000-00805f9b34fb}	System.Guid

            var stringToSend = "Hello World!:) $" + Environment.NewLine;
            
            var services = await deviceConn.GetServicesAsync();

            service = await deviceConn.GetServiceAsync(Guid.Parse("0000ffe0-0000-1000-8000-00805f9b34fb"));
            
            var characteristics = await service.GetCharacteristicsAsync();
            characteristic = await service.GetCharacteristicAsync(Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb"));

            
           
            try
            {
                byte[] receivedBytes;

                //Write message
                byte[] bytes = Encoding.ASCII.GetBytes(stringToSend);

                var result = await characteristic.WriteAsync(bytes);

                msgSended = "Message sended:" + stringToSend;

                //Receive message

                 characteristic.ValueUpdated += async (o, args) =>
                 {
                     receivedBytes = args.Characteristic.Value;
                     mesRecBT = Encoding.ASCII.GetString(receivedBytes, 0, receivedBytes.Length);

                 };
                
                await characteristic.StartUpdatesAsync();

                msgReceived = "Msg received: " + mesRecBT;

            }
            catch (Exception ex)
            {
                Console.WriteLine(ex);
            }
            
        }
```
When I capture characteristics by GetCharacteristicsAsync(), only appears one of them, so I do not how to separate Read/Write using characteristics.

andrey-val-rodin, 2022-10-12:
Hi Raul,

 

>>but my BLE module does not offer two different characteristics

 

Are you sure? I worked with module HC-6. It has two characteristics in service 0000ffe0-0000-1000-8000-00805f9b34fb:

0000ffe1-0000-1000-8000-00805f9b34fb – can be used for reading (update) on app side

0000ffe2-0000-1000-8000-00805f9b34fb – can be used for writing

 

Try the following code to receive data on Arduino side:

 

String buffer;

void loop()

{

  if (BT.available() > 0)

  {

    buffer += BT.readStringUntil('\n');

  }

 

  if (buffer.endsWith("\n"))

  {

    Serial.print(buffer);

    // Your logic here

    // ...

    // Clear buffer

    buffer = "";

  }

}

 

Regarding strict and professional rules: ok, but you should remember that characteristic.StartUpdatesAsync() and characteristic.ValueUpdated += async (o, args) => {…} must be called only once at the time of initialization.

 

Regards,

Andrey

 

 <https://user-images.githubusercontent.com/91801890/195386148-36628b3f-bd55-4711-838e-5c4d5263e0c6.PNG>

rauleteee, 2022-10-12:
Nope, not sure. If I debug, I only see one of the characteristics on the List:
[image]
So, that is why I really did not separate write/read functions (it was my first idea).

I'll try your buffer idea on arduino, thank you very much!

rauleteee, 2022-10-12:
That is what I received if I try to send string with more than 20 chars, even clearing the buffer with the end of the string:
[image]

andrey-val-rodin, 2022-10-12:
So now everything is ok on Arduino side?

 

I suppose you get only final part of the message on app side because you set event and then call StartUpdatesAsync too late. Move initialization to separate function and call it before sending string. Use the same trick for buffering input as in Arduino.

 

Using the same characteristic for reading and writing is not a problem in your case, I think, because you are not going to read and write simultaneously.

 

Andrey.

 

 <https://user-images.githubusercontent.com/91801890/195401590-f59edaf7-f99b-45df-a2df-1a4a6147c9b4.PNG>

rauleteee, 2022-10-12:
Hi Audrey,

So I move all the device connection/services/characteristics initialization to the start of the app, characteristic.ValueUpdated += (o, args) ..., and characteeristic.StartUpdateASync(). Now what happens is that I receive the first 20B of the string,and nothing else. Here is a snap of the app:
[image]

I do the same as in the arduino code, saving the string to a buffer as follows:
```
 public async Task sendString()
        {
            //Guid characteristics = 	Id	{0000ffe1-0000-1000-8000-00805f9b34fb}	System.Guid
            //Guid services = 	Id	{0000ffe0-0000-1000-8000-00805f9b34fb}	System.Guid

            var stringToSend = "Large message to the bluetooth" + Environment.NewLine;
 
           
            try
            {
                

                //Write message
                byte[] bytes = Encoding.ASCII.GetBytes(stringToSend);

                var result = await characteristic.WriteAsync(bytes);

                msgSended = "Message sended:" + stringToSend;

                //Receive message

                var buff = await characteristic.ReadAsync();
                var stringBuff = Encoding.ASCII.GetString(buff);
                char v = stringBuff.Last();
                if (v.Equals(Environment.NewLine))
                    stringBuff = "";

                mesRecBT += stringBuff;

                msgReceived = "Msg received: " + mesRecBT;

            }
            catch (Exception ex)
            {
                Console.WriteLine(ex);
            }
            
        }
```
I tried to display on Serial Monitor of arduino what bluetooth catches, and it takes all the large string. So at least I know that the error is occurring on xamarin client.

Any idea why is the xamarin client cutting the string?

Thank you in advance.

Regards,
Raúl.

rauleteee, 2022-10-12:
[image]

Serial monitor snap when I changed BT.print(buffer) to Serial.print(buffer).

rauleteee, 2022-10-12:
Not only the problem of shorten string, but also the buffer does not change anymore if I move characteristic.ValueUpdated to the initialization.
[image]

andrey-val-rodin, 2022-10-12:
I don't advise you to use characteristic.ReadAsync(). And don't confuse Read and Update. When you use Update technique, your event handler will be invoked every time for a new incoming part of the string (or once for a short string). Just gather all buffering logic in event handler. Specify buffer as class member. The presence of a terminator at the end of the buffer will mean that data transfer process is completed.

rauleteee, 2022-10-12:
Sorry for bothering, I am really trying to understand what is happening. Now this is my constructor on xamarin client: 
```
        #region CONSTRUCTOR
        //InitializeComponent()
        public PostViewModel(INavigation navigation)
        {
            initializeBluetooh();

            //>>>>>>>>>Para que pueda ir hacia delante y hacia detrás
            Navigation = navigation;
        }
        async void initializeBluetooh()
        {
            _handler = CrossBluetoothLE.Current;
            _adapter = CrossBluetoothLE.Current.Adapter;

            devicelist = new ObservableCollection<IDevice>();

            guid = new Guid("96af0bb8-cafa-5b07-a0d2-79c01a414e7e");

            if (!_handler.Adapter.IsScanning)
            {
                devicelist.Clear();

                try
                {
                    deviceConn = await _adapter.ConnectToKnownDeviceAsync(guid, new Plugin.BLE.Abstractions.ConnectParameters(), new CancellationToken());
                }
                catch (DeviceConnectionException ex)
                {
                    Console.WriteLine(ex.ToString());
                }

            }
            bt_state = "BLUETOOTH INFO:\n" +
                "\nBluetooth state: " + deviceConn.State +
                "\nConnected to: " + deviceConn.Name;

            services = await deviceConn.GetServicesAsync();

            service = await deviceConn.GetServiceAsync(Guid.Parse("0000ffe0-0000-1000-8000-00805f9b34fb"));

            characteristics = service.GetCharacteristicsAsync();
            characteristic = await service.GetCharacteristicAsync(Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb"));
            

            characteristic.ValueUpdated += async (o, args) =>
            {

                byte[] receivedBytes = args.Characteristic.Value;
                stringBuff = Encoding.ASCII.GetString(receivedBytes, 0, receivedBytes.Length);
                if (stringBuff.EndsWith(Environment.NewLine))
                    stringBuff = "";

            };

            await characteristic.StartUpdatesAsync();

        }
```

Send string function:
```
 public async Task sendString()
        {
            //Guid characteristics = 	Id	{0000ffe1-0000-1000-8000-00805f9b34fb}	System.Guid
            //Guid services = 	Id	{0000ffe0-0000-1000-8000-00805f9b34fb}	System.Guid

            var stringToSend = "Hello world :)" + Environment.NewLine;

            try
            { 

                //Write message
                byte[] bytes = Encoding.ASCII.GetBytes(stringToSend);

                var result = await characteristic.WriteAsync(bytes);
                

                msgSended = "Message sended:" + stringToSend;
                //Thread.Sleep(1000);
                

                msgReceived = "Msg received: " + stringBuff;

            }
            catch (Exception ex)
            {
                Console.WriteLine(ex);
            }
            
        }
```
And arduino code:
```
void loop(){
  // recibe mensaje Bluetooth y envia por USB a PC
  if (BT.available() > 0 ){
    
   buffer += BT.readStringUntil('\n');
    BT.print(buffer);
  }
  if(buffer.endsWith("\n")){
    buffer = "";

  }
  }
```
App's behaviour:

https://user-images.githubusercontent.com/91801890/195439997-681801a3-e5a1-4e69-a062-3787f6d7cd5e.MP4

rauleteee, 2022-10-12:
Sorry for bothering, I am really trying to understand what is happening. Now this is my constructor on xamarin client:
```
        #region CONSTRUCTOR
        //InitializeComponent()
        public PostViewModel(INavigation navigation)
        {
            initializeBluetooh();

            //>>>>>>>>>Para que pueda ir hacia delante y hacia detrás
            Navigation = navigation;
        }
        async void initializeBluetooh()
        {
            _handler = CrossBluetoothLE.Current;
            _adapter = CrossBluetoothLE.Current.Adapter;

            devicelist = new ObservableCollection<IDevice>();

            guid = new Guid("96af0bb8-cafa-5b07-a0d2-79c01a414e7e");

            if (!_handler.Adapter.IsScanning)
            {
                devicelist.Clear();

                try
                {
                    deviceConn = await _adapter.ConnectToKnownDeviceAsync(guid, new Plugin.BLE.Abstractions.ConnectParameters(), new CancellationToken());
                }
                catch (DeviceConnectionException ex)
                {
                    Console.WriteLine(ex.ToString());
                }

            }
            bt_state = "BLUETOOTH INFO:\n" +
                "\nBluetooth state: " + deviceConn.State +
                "\nConnected to: " + deviceConn.Name;

            services = await deviceConn.GetServicesAsync();

            service = await deviceConn.GetServiceAsync(Guid.Parse("0000ffe0-0000-1000-8000-00805f9b34fb"));

            characteristics = service.GetCharacteristicsAsync();
            characteristic = await service.GetCharacteristicAsync(Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb"));
            

            characteristic.ValueUpdated += async (o, args) =>
            {

                byte[] receivedBytes = args.Characteristic.Value;
                stringBuff = Encoding.ASCII.GetString(receivedBytes, 0, receivedBytes.Length);
                if (stringBuff.EndsWith(Environment.NewLine))
                    stringBuff = "";

            };

            await characteristic.StartUpdatesAsync();

        }
```
Send string function:
```
 public async Task sendString()
        {
            //Guid characteristics = 	Id	{0000ffe1-0000-1000-8000-00805f9b34fb}	System.Guid
            //Guid services = 	Id	{0000ffe0-0000-1000-8000-00805f9b34fb}	System.Guid

            var stringToSend = "Hello world :)" + Environment.NewLine;

            try
            { 

                //Write message
                byte[] bytes = Encoding.ASCII.GetBytes(stringToSend);

                var result = await characteristic.WriteAsync(bytes);
                

                msgSended = "Message sended:" + stringToSend;
                //Thread.Sleep(1000);
                

                msgReceived = "Msg received: " + stringBuff;

            }
            catch (Exception ex)
            {
                Console.WriteLine(ex);
            }
            
        }
```
And arduino code:
```
void loop(){
  // recibe mensaje Bluetooth y envia por USB a PC
  if (BT.available() > 0 ){
    
   buffer += BT.readStringUntil('\n');
    BT.print(buffer);
  }

https://user-images.githubusercontent.com/91801890/195440130-cb4e5337-4f3d-402a-a3a6-20cb4f34b91f.mov

  if(buffer.endsWith("\n")){
    buffer = "";

  }
  }
```
App's behaviour:

https://user-images.githubusercontent.com/91801890/195440355-de7c6949-45eb-4716-b4a7-cb46bc3ed6ef.mov

I really understand that the value is updating all the time when StartUpdateASync is called on Event function at the start of the app, but I do not understand why the value received is random.

Regards,
Raúl.

rauleteee, 2022-10-12:
Thank you for your patience, I finally find the solution. I was saving the buffer as buffer= ...GetString() and the good way was buffer+=...GetString(). 

Best regards,
Raúl.

closed 2022-10-12 by rauleteee