dotnet-bluetooth-le/dotnet-bluetooth-le #630: MauiXamlBluetoothLe builds on Mac but only runs for Android platform.

jdg3k92, 2022-10-14:
## Steps to reproduce

1. Install Visual Studio 2022 for Mac (17.3.7)

2. Clone THIS repo and open in VS.

3. Build and run on ios simulator

## Expected behavior
The App should load and display the landing page in iOS Simulator (Works fine for Android Simulator and Real Android device)

## Actual behavior
code blows up and throws exception. Main message is the following: 

This functionality is not implemented in the portable version of this assembly.  You should reference the NuGet package from your main application project in order to reference the platform-specific implementation.

### Crashlog

   at Plugin.BLE.CrossBluetoothLE.get_Current()
   at MauiXAMLBluetoothLE.Services.BluetoothLEService..ctor() in /Users/jgeshrick/Projects/MauiXAMLBluetoothLE-master/MauiXAMLBluetoothLE/Services/BluetoothLEService.cs:line 13
   at System.Reflection.RuntimeConstructorInfo.InternalInvoke(Object obj, Object[] parameters, Boolean wrapExceptions)
   at System.Reflection.RuntimeConstructorInfo.DoInvoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitConstructor(ConstructorCallSite constructorCallSite, RuntimeResolverContext context)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2[[Microsoft.Extensions.DependencyInjection.ServiceLookup.RuntimeResolverContext, Microsoft.Extensions.DependencyInjection, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60],[System.Object, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].VisitCallSiteMain(ServiceCallSite callSite, RuntimeResolverContext argument)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitRootCache(ServiceCallSite callSite, RuntimeResolverContext context)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2[[Microsoft.Extensions.DependencyInjection.ServiceLookup.RuntimeResolverContext, Microsoft.Extensions.DependencyInjection, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60],[System.Object, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].VisitCallSite(ServiceCallSite callSite, RuntimeResolverContext argument)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitConstructor(ConstructorCallSite constructorCallSite, RuntimeResolverContext context)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2[[Microsoft.Extensions.DependencyInjection.ServiceLookup.RuntimeResolverContext, Microsoft.Extensions.DependencyInjection, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60],[System.Object, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].VisitCallSiteMain(ServiceCallSite callSite, RuntimeResolverContext argument)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitRootCache(ServiceCallSite callSite, RuntimeResolverContext context)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2[[Microsoft.Extensions.DependencyInjection.ServiceLookup.RuntimeResolverContext, Microsoft.Extensions.DependencyInjection, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60],[System.Object, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].VisitCallSite(ServiceCallSite callSite, RuntimeResolverContext argument)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitConstructor(ConstructorCallSite constructorCallSite, RuntimeResolverContext context)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2[[Microsoft.Extensions.DependencyInjection.ServiceLookup.RuntimeResolverContext, Microsoft.Extensions.DependencyInjection, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60],[System.Object, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].VisitCallSiteMain(ServiceCallSite callSite, RuntimeResolverContext argument)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.VisitRootCache(ServiceCallSite callSite, RuntimeResolverContext context)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteVisitor`2[[Microsoft.Extensions.DependencyInjection.ServiceLookup.RuntimeResolverContext, Microsoft.Extensions.DependencyInjection, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60],[System.Object, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].VisitCallSite(ServiceCallSite callSite, RuntimeResolverContext argument)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.CallSiteRuntimeResolver.Resolve(ServiceCallSite callSite, ServiceProviderEngineScope scope)
   at Microsoft.Extensions.DependencyInjection.ServiceProvider.CreateServiceAccessor(Type serviceType)
   at System.Collections.Concurrent.ConcurrentDictionary`2[[System.Type, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.Func`2[[Microsoft.Extensions.DependencyInjection.ServiceLookup.ServiceProviderEngineScope, Microsoft.Extensions.DependencyInjection, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60],[System.Object, System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], System.Private.CoreLib, Version=6.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].GetOrAdd(Type key, Func`2 valueFactory)
   at Microsoft.Extensions.DependencyInjection.ServiceProvider.GetService(Type serviceType, ServiceProviderEngineScope serviceProviderEngineScope)
   at Microsoft.Extensions.DependencyInjection.ServiceLookup.ServiceProviderEngineScope.GetService(Type serviceType)
   at Microsoft.Maui.MauiContext.WrappedServiceProvider.GetService(Type serviceType)
   at Microsoft.Maui.Controls.ShellContent.<>c__DisplayClass19_0.<Microsoft.Maui.Controls.IShellContentController.GetOrCreateContent>b__0()
   at Microsoft.Maui.Controls.ElementTemplate.CreateContent()
   at Microsoft.Maui.Controls.Internals.DataTemplateExtensions.CreateContent(DataTemplate self, Object item, BindableObject container)
   at Microsoft.Maui.Controls.ShellContent.Microsoft.Maui.Controls.IShellContentController.GetOrCreateContent()
   at Microsoft.Maui.Controls.Platform.Compatibility.ShellSectionRootRenderer.LoadRenderers()
   at Microsoft.Maui.Controls.Platform.Compatibility.ShellSectionRootRenderer.ViewDidLoad()
   at UIKit.UIApplication.Main(String[] args, Type principalClass, Type delegateClass)
   at MauiXAMLBluetoothLE.Program.Main(String[] args) in /Users/jgeshrick/Projects/MauiXAMLBluetoothLE-master/MauiXAMLBluetoothLE/Platforms/iOS/Program.cs:line 13

## Configuration

**Version of the Plugin:**  2.1.3

**Platform:** e.g. iOS 15.5 / Android 12

[image]

[image]

tmuskopf, 2022-10-23:
I am facing this exact issue with my application. I am very interested in this solution. Only hitting android users for beta testing will limit the usefulness of my beta testing!

janusw, 2022-10-27:
I have not had a look at MauiXamlBluetoothLe yet.

In general it should be noted that this library does not support MAUI yet, but it should once #614 is merged.

janusw, 2022-11-16:
This problem should hopefully be fixed by https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/638.

@jdg3k92 @TimoKinnunen, can you confirm that the problem is gone on current master? (using the nupkg from here: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/3470893880#artifacts)

TimoKinnunen, 2022-11-16:
I have acquired iPhone 10 but lacking Apple developer account can only test for Android and Windows. I see how hard you are solving and developing and want to say thank you!

janusw, 2022-11-16:
> I have acquired iPhone 10 but lacking Apple developer account can only test for Android and Windows.

Well, yeah. The Apple universe is not exactly developer friendly. Everything is very much locked-in, and you basically cannot do anything without Apple's approval. Anyway, setting up a basic developer account is free, after all, so maybe you might create one anyway.

In any case, maybe @jdg3k92 can help out here? (If not, I might try to look into it soon.)

TimoKinnunen, 2022-11-16:
I will buy Apple developer account soon.  What can I do with [dotnet-bluetooth-le](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le)? I have code in Visual Studio but what then?

tmuskopf, 2022-11-16:
I am also close to deploying the latest nuget from: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/3470893880#artifacts
I just got my apple developer account activated (it takes 48 hours from purchase😒). 
My app scans, connects, and subscribes to ble notify events and receives raw data packets. Fully functional on Android.
I'll update here when I get some time to do this integration and testing.

janusw, 2022-11-17:
> I will buy Apple developer account soon.

In fact it's not free (contrary to what I claimed above), but for simple on-device testing you might even get away with an Apple ID only (which is free).

> What can I do with [dotnet-bluetooth-le](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le)? I have code in Visual Studio but what then?

Sorry, I don't understand the question? What do you mean?

janusw, 2022-11-17:
> My app scans, connects, and subscribes to ble notify events and receives raw data packets. Fully functional on Android. I'll update here when I get some time to do this integration and testing.

@tmuskopf, yes, please let us know how it goes ...

TimoKinnunen, 2022-11-17:
I don't know how to make Plugin.BLE from code. I use Plugin.BLE in my Visual Studio solution. I get Plugin.BLE from NuGet as package. This is new to me.

janusw, 2022-11-17:
> I don't know how to make Plugin.BLE from code. I use Plugin.BLE in my Visual Studio solution. I get Plugin.BLE from NuGet as package.

Actually you don't need to build a nupkg yourself, you can just download the one from this link:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/3470893880#artifacts
Then you can simply put Plugin.BLE.2.2.0-pre5-34-g17221bc.nupkg into some folder on your build machine, and configure that folder as a "local package source" in VS. After that you should be able to update to that version via the VS package manager.

janusw, 2022-11-17:
Actually by now I was able to verify myself that the problem is gone. With Plugin.BLE.2.2.0-pre5-23-g0f2abde, I still get the exception, but with Plugin.BLE.2.2.0-pre5-34-g17221bc it is gone and MauiXamlBluetoothLe starts up without error on my iPhone 🎉 

Therefore I'll close this issue.

TimoKinnunen, 2022-11-17:
I have hope. Thank you Janus Weil for hard work.