dotnet-bluetooth-le/dotnet-bluetooth-le #635: Sequential writes breaks characteristic.ValueUpdated event

RobertLinak, 2022-11-03:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

## Configuration

**Version of the Plugin:** 2.2.0-pre2 to 2.2.0-pre5

**Platform:** Android 11.0 - API 30 

**Device:** Samsung SM-A325F

janusw, 2022-11-03:
See also the discussion in PR #614, in particular:
* https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/614#issuecomment-1300469397
* https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/614#issuecomment-1301841247

janusw, 2022-11-03:
Looking into the git history between 2.2.0-pre1 (working) and 2.2.0-pre2 (not working), I suspect that this might be related to #376. @RobertLinak, do you think this could be the culprit? The relevant commits are 4fd481f4462a25b96492dfca799c4f3bf6603d02, 0447b9feffe730034c93aac925f69768709f1d69 and  5f41b691447992eac9e3d2abf8f2ae4fe5c88399.

I haven't looked into it very deeply, but possibly it's more of a change of behavior than an actual bug?

RobertLinak, 2022-11-03:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/276#issuecomment-581126374

RobertLinak, 2022-11-03:
> Looking into the git history between 2.2.0-pre1 (working) and 2.2.0-pre2 (not working), I suspect that this might be related to #376. @RobertLinak, do you think this could be the culprit? The relevant commits are [4fd481f](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/4fd481f4462a25b96492dfca799c4f3bf6603d02), [0447b9f](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/0447b9feffe730034c93aac925f69768709f1d69) and [5f41b69](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/5f41b691447992eac9e3d2abf8f2ae4fe5c88399).
> 
> I haven't looked into it very deeply, but possibly it's more of a change of behavior than an actual bug?

Yes this definitely looks like it could be the reason.
Would be nice if this experimental feature could be enabled/disabled as stated in #376 but I dont see anything relevant in the sample project and I am not able see the code changes made in 2.2.0-pre2 either.

janusw, 2022-11-03:
> [#276 (comment)](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/276#issuecomment-581126374) So this looks to be the exact problem Im having, and it looks to be ignored since 2020 from what I can see.

The problem here is that @xabre basically retired (or paused?) from being the lead maintainer of this repo, and the changes for #376 were among the last things he did (before he practically disappeared).

RobertLinak, 2022-11-03:
So I found this https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/4fd481f4462a25b96492dfca799c4f3bf6603d02 

///summary
/// Platform specific main thread invocation. Useful to avoid GATT 133 errors on Android.
/// Set this to NULL in order to disable main thread queued invocations.
/// Android: already implemented and set by default
/// UWP, iOS, macOS: NULL by default - not needed, turning this on is redundant as it's already handled internaly by the platform
/// summary
public static Action<Action> MainThreadInvoker { get; set; }

So I tried setting
Plugin.BLE.Abstractions.Utils.TaskBuilder.MainThreadInvoker = null;
But it does not solve the problem.

closed 2022-11-03 by RobertLinak

janusw, 2022-11-03:
> So I tried setting Plugin.BLE.Abstractions.Utils.TaskBuilder.MainThreadInvoker = null;
> 
> ... This actually does solve the problem when doing it after a connection is made to a device. So I guess that is the solution to this issue then.

Awesome, great to hear that 👍  🎉