dotnet-bluetooth-le/dotnet-bluetooth-le #641: .NET Maui and Plugin.BLE 3.0.0-beta.2. Plugin.BLE does not detect Bluetooth LE devices on Windows. Seems this is a limitation of the current implementation for Windows.

TimoKinnunen, 2022-11-24:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

Youldboukhitine, 2022-11-27:
Tested on my side with .NET MAUI windows  with the same release no major issue. 
Working well.

TimoKinnunen, 2022-11-27:
My mistake!!!! I made a new test and it works.
Polar H7 Heart Rate Sensor (Bluetooth LE device) on Windows is detected by Plugin.BLE version 3.0.0-beta.2.
I am pleased. Thank you Janus Weil so far.
Android is ok. Windows is ok. iOS I don't know because I haven't tested it on my iPhone 10 yet. But I believe what you said that it is ok. Wonderful news.

I close this issue now.

closed 2022-11-27 by TimoKinnunen