dotnet-bluetooth-le/dotnet-bluetooth-le #642: [MAUI - WINDOWS ] BluetoothLE attributes IsOn and IsAvailable return always true independant of the Bluetooth state on the device.

Youldboukhitine, 2022-11-27:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

The BluetoothLE attributes IsOn and isAvailable are not matching with the device state (computer bluetooth peripheral).
These attributes are always to true. 

Is there any other way to check if the Bluetooth peripheral is enabled on the device side ?

MAUI with .NET 7 on Windows 10.x

## Steps to reproduce

1. On windows -- > Settings --> Bluetooth and other devices

2. Turn OFF Bluetooth

3. create a simple app and read the status of a BluetoothLE  variable 

## Expected behavior
The attributes IsOn and IsAvailable shall match with the Bluetooth peripheral state

## Actual behavior
got always true

### Crashlog

N/A

## Configuration

**Version of the Plugin:** 3.0.0-beta-2

**Platform:** Windows 10 OS build 19044.2130

**Device:** : DELL Laptop

janusw, 2022-11-27:
See `BleImplementation.GetInitialStateNative`: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/f474528254bf227dc984477a9367aa088d17c301/Source/Plugin.BLE/Windows/BleImplementation.cs#L29-L40

I guess one should at least check [BluetoothLEHelper.IsBluetoothLESupported](https://learn.microsoft.com/en-us/dotnet/api/microsoft.toolkit.uwp.connectivity.bluetoothlehelper.isbluetoothlesupported?view=win-comm-toolkit-dotnet-6.1#microsoft-toolkit-uwp-connectivity-bluetoothlehelper-isbluetoothlesupported) in addition.

Youldboukhitine, 2022-11-27:
@janusw Thanks for the info.

janusw, 2022-11-28:
Why did you close this issue? It's a valid bug report, and I simply wanted to point out the relevant piece of code. Despite the comment there, I think the behavior can and should be improved ...

Youldboukhitine, 2022-11-28:
I don't to waste the repo with something that you already considered. 
That's ok for me to leave it open :)

DamianSuess, 2023-01-29:
As I too ran into the same issue (again) with the library, the missing UWP feature could no longer be ignored. So I went ahead and made a branch which implements UWP's [Radio.StateChanged Event](https://learn.microsoft.com/en-us/uwp/api/windows.devices.radios.radio.statechanged?view=winrt-19041). Just to call out a couple of obvious minor hurdles w/ the project that I believe need to be discussed in the near future.

@janusw you already mentioned the `async` one. I had to do a minor workaround for it.. though I'm sure there's a better way.

The next is that the Community Toolkit exposes only the default BLE Adapter. It would be better if the collection adapters was returned rather than using, `BluetoothAdapter.GetDefaultAsync()`. This would pair well with the `Radio.StateChanged` event.

youldb, 2023-01-29:
Nice work !
Do you think that the task waiting time has a dependency with hardware driver ?
If so we might get some troubles if some devices react with different time ?

DamianSuess, 2023-01-30:
Overall, the user can always re-pull the status if need be. It was observed that it only affected the initial value of the IsAvailable status. Open to suggestions

janusw, 2023-10-31:
Could someone check if this problem is fixed by #746?

Auto72, 2024-01-15:
Time ago I solved this way for Windows.

#if WINDOWS
        blePresent = await BleAvailableAsync();
        bleEnabled = awaitBleEnabledAsync();
#endif

#if WINDOWS
    public static async Task<bool> BleAvailableAsync()
    {
        var radios = await Radio.GetRadiosAsync();
        return radios.FirstOrDefault(radio => radio.Kind == RadioKind.Bluetooth) != null;
    }

    public static async Task<bool> BleEnabledAsync()
    {
        var radios = await Radio.GetRadiosAsync();
        var bluetoothRadio = radios.FirstOrDefault(radio => radio.Kind == RadioKind.Bluetooth);
        return bluetoothRadio != null && bluetoothRadio.State == RadioState.On;
    }
#endif

AskBojesen, 2024-02-05:
@Youldboukhitine: Have you tested with latest in master?
I cannot reproduce - I believe this issue is resolved and can be closed

janusw, 2024-02-05:
Alright, then let's close it. If this or a similar problem still occurs, please open a new issue.

closed 2024-02-05 by janusw

Auto72, 2024-02-06:
Sorry, the following code on Windows doesn't work if the USB BLE Adapter is unplugged:
(you can add this code to your BLE.Client.WinConsole project in the file program.cs)

var blePresent = CrossBluetoothLE.Current.IsAvailable;
var bleEnabled = CrossBluetoothLE.Current.IsOn;

Console.WriteLine($"\nCrossBluetoothLE.Current.IsAvailable: {blePresent} | CrossBluetoothLE.Current.IsOn: {bleEnabled}\n");

You get the error reported below in the picture.

[image]

Above in this thread I provided the working code.