dotnet-bluetooth-le/dotnet-bluetooth-le #645: Mac Address issue in IOS

LiftTrack, 2022-11-28:
Hi,

I am using Plugin.BLE, I am facing one issue, I am unable to get Mac Id of the discovered devices by advertisements. Amazing fact is i am able to get Mac address of discovered devices in Android. I am using same plugin for both. can you help me in this issue or suggest me how can i get Mac address of discovered devices in IOS xamarin forms. In Android version, it is available in "e.Device.NativeDevice" which I get from advertisements but in IOS plugin i am unable to get Mac address of discovered devices.

Please help me in it, if there is any solution for it.

smstuebe, 2022-11-28:
AFAIK you can't get the MAC Addresses in iOS. 
https://developer.apple.com/forums/thread/8442

If you really need the Mac Address, you need to put it in the Advertisement as payload (obv. only works if you develop the device itself, too).

janusw, 2022-11-30:
> AFAIK you can't get the MAC Addresses in iOS. https://developer.apple.com/forums/thread/8442

Yes, this is simply not possible on iOS (platform limitation). Unfortunately raising three issues at once does not help either ;)

In any case, this is pretty much an FAQ, so maybe we should actually set up an FAQ page to clarify such questions ...?

closed 2022-11-30 by janusw