dotnet-bluetooth-le/dotnet-bluetooth-le #646: Xamarin.Forms UWP DeviceConnected сalled multiple times

vitalii-vov, 2022-12-09:
Hello.
After disconnecting from the device and reconnecting, the DeviceConnected callback will be called twice. If you connect for the third time, then the DeviceConnected callback will be called three times, and so on. 
In this case, DeviceDisconnected is called once.
On android it works without errors.

## Steps to reproduce

1. call ```await _adapter.ConnectToDeviceAsync(_device);```
```DeviceConnected``` called once
2. call ```await _adapter.DisconnectDeviceAsync(_device);```
3. call ```await _adapter.ConnectToDeviceAsync(_device);```
```DeviceConnected``` called twice

## Expected behavior
```DeviceConnected``` called once all time

## Actual behavior
```DeviceConnected``` is called as many times as we reconnect to the device

## Configuration
**Version of the Plugin:** 3.0.0-beta.2
**Platform:** Windows 10 Pro 19045.2251

Demonstrate project
[BleBug.zip](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/files/10193723/BleBug.zip)

janusw, 2022-12-10:
There are already a couple of issues that concern disconnection and reconnection problems on UWP:
* #423
* #528
* #536
* #620

Could they be related to your problem? Do they still occur? Or are they fixed?

Does the disconnection and reconnection actually work in your setup?

vitalii-vov, 2022-12-10:
I use this library to work on ios and android and it works great. For UWP projects, I use a different library, so I don't have much information about UWP bugs. But, I wanted to completely switch over to this library, because its design is better. That's why I started testing it for UWP, and it really has problems with disconnecting as well as connecting.

If we disconnect from the device, wait a bit, and reconnect, then the next attempt to disconnect the device will keep the device connected on UWP.

For example, when we need to reconnect to the device at the moment when we put it into OTA mode, in case of a firmware update.

```csharp
async Task Foo() {
    //  ...
    // We are connected to the device before
    //  We send a command to the device to switch to firmware mode.
    //  We need to reconnect to the device before starting to write new firmware to it and in order to get new characteristics.
    await ReconnectToOtaDeviceAsync();

    //  ... 
    //  We send the firmware to the device.
    // ..

    //  Try to disconnect
    await _adapter.DisconnectDeviceAsync(_device); //  <- Device is still connected 
    _device.Dispose();
}

private async Task ReconnectToOtaAsync()
{
    await Task.Delay(5000);
    var deviceUUID = _device.Id;
    await _adapter.DisconnectDeviceAsync(_device);
    _device = await _adapter.ConnectToKnownDeviceAsync(deviceUUID);
    await Task.Delay(2000);
}
```

Also, if the device is turned off manually by pressing the power button, the disconnect event from the device will be triggered, but when you try to reconnect to the device, it will no longer connect.