dotnet-bluetooth-le/dotnet-bluetooth-le #647: Plugin.BLE.Abstractions.Exceptions.DeviceConnectionException: GattCallback error: 133   at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System

BasuHuddar, 2022-12-12:
Hi,

I'm using Plugin.BLE 2.1.3 for my project but I get an exception when I try to connect my phone to a BLE device.

Steps to reproduce

I scan for all of the devices
I choose the device I want to connect it, but after 10 to 20 seconds getting a connection exception as GatCallback error: 133
async void Connection(object sender, EventArgs e)
{
    try
        {
            await App.Adapter.ConnectToDeviceAync(_connectedDevice);
        }
        catch (DeviceConnectionException e){
        }
    }
}
Expected behavior

Connection should succeed.

Actual behavior

I got a DeviceConnectionException with this message

GattCallback error: 133
Configuration

**Version:**
Visual Studio Community 2019 for Mac
Version 8.10.25 (build 2)
Xamarin.Forms: 5.0.0.2515
Plugin.BLE 2.1.3

oguzycl, 2023-02-21:
Hi,

Did you find any solution for this error. I am using code snipped on the below for connecting ble device my android phone and i got same error message. 

```
private async Task<(bool, IDevice)> ConnectTagAsync(Tags tag)
        {
            var currentTag = tag.TagDev;
            if (currentTag == null)
            {
                Console.WriteLine("tag is null");
                await UserDialogs.Instance.AlertAsync("Error. No connectable Tag");
                return (false,currentTag);
            }
            if(_adapter == null)
            {
                Console.WriteLine("adapter is null");
                await UserDialogs.Instance.AlertAsync("Error. Adapter not found");
                return (false, currentTag);
            }
            try
            {
                await _adapter.ConnectToDeviceAsync(currentTag);
                return (true, currentTag);
            }
            catch(Exception ex)
            {
                Console.WriteLine($"didnt connect to device exception: {ex}");
                await UserDialogs.Instance.AlertAsync($"Connection fail. Error massage: {ex}");
                return (false, currentTag);
            }
        }
```

92-Theo, 2023-02-21:
My app is also has users occur Gatt Status 133 or 257 after Android 13 update.
And users sholud reboot the phone.

Someone sent error to Samsung Phone.
And we got error message and solution from Samsung.
Get help from your phone manufacturer.

Please share issue if you have some solution.
Thanks.

```
 12-27 18:33:08.061--gatt_api.cc -- can't Register GATT client, MAX client reached: 39
   12-27 18:33:18.044--gatt_api.cc -- can't Register GATT client, MAX client reached: 39
   12-27 18:33:18.044--gatt_api.cc -- can't Register GATT client, MAX client reached: 39
   12-27 18:33:18.045--gatt_api.cc -- can't Register GATT client, MAX client reached: 39

   appIf: 1, ..
   appIf: 2, ..
   appIf: 3, ..
   appIf: 4, ..
   appIf: 5, ..
   appIf: 6, ..
   appIf: 7, ..
   appIf: 8, ..
   appIf: 9, ..
   appIf: 28, ..

   There may be Bluetooth GATT resource wasting app(s).
   App should unregister itself after using Bluetooth resource. But, this app doesn't unregister after using and register again.
   Please remove this app and see what happens.
```

RAMESHKUMAR502, 2023-08-29:
Hi All,
We are able to see the same issue.do we have any solution? For the Below error
System.Exception: Device '00000000-0000-0000-0000-3414b58f1f71' disconnected while writing descriptor with 00002902-0000-1000-8000-00805f9b34fb.