dotnet-bluetooth-le/dotnet-bluetooth-le #649: Maui NET 7.0 Plugin.BLE through System.NotImplementException in WIndows Machine

nefen, 2023-01-07:
Installing latest Visual Studio (17.4.3)

Start a Maui Project on .NET 7.0.

Install PLugin.BLE latest version (2.1.3)

Android is ok, but Windows Machine through a 'System.NotImplementedException' in Plugin.BLE.dll

I searched that we also need Microsoft.Toolkit.Uwp.Connectivity to be installed, although this package is not compatible with .NET 7.0 Maui.

DasVerma, 2023-01-07:
Hi, you need to install 3.0.0-beta.2 version to support uwp and then
Microsoft.Toolkit.Uwp.Connectivity(>7.1.2)

nefen, 2023-01-07:
I installed 3.0.0-beta.2.
Microsoft toolkit requires max windows 10.0.17763, but Microsoft.Maui.Graphics is blocked to downgrade from the project at Net 7.0

janusw, 2023-01-07:
> Microsoft toolkit requires max windows 10.0.17763

I think Microsoft.Toolkit.Uwp.Connectivity requires **at least** Windows 10.0.17763 (min not max).

Microsoft.Maui.Graphics 7.x apparently requires at least 10.0.19041, so targeting 19041 should satisfy both of them, no?

> but Microsoft.Maui.Graphics is blocked to downgrade from the project at Net 7.0

I don't quite understand what you're saying here, but I can add net7.0 targets to Plugin.BLE (if it helps). 3.0.0-beta.2 currently only targets net6.0.

janusw, 2023-01-07:
I'm adding .NET 7 support in #650.

@nefen Could you check if the nupkg from https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/3862299303#artifacts works for you?

nefen, 2023-01-07:
@janusw thank you very much for your effort and prompt reply!
Although you say that 3.0.0.beta.2 didn't support Maui .NET 7, I installed nuget package and my code worked for android.
On WIndows Machine I faced the notImplement error and found that I need to install Microsoft.Toolkit.Uwp.Connectivity, which does not install as it is not compatible for versions ABOVE 10.0.17763:
Severity	Code	Description	Project	File	Line	Suppression State
Error	NU1202	Package Microsoft.Toolkit.Uwp.Connectivity 7.1.3 is not compatible with net7.0-android33.0 (.NETCoreApp,Version=v7.0) / win10-x64. Package Microsoft.Toolkit.Uwp.Connectivity 7.1.3 supports: uap10.0.17763 (UAP,Version=v10.0.17763)	efce3D_ble	D:\Development\efce3d-artemis\efce3D_ble_artemis\efce3D_ble\efce3D_ble.csproj	
Error	NU1202	Package Microsoft.Toolkit.Uwp.Connectivity 7.1.3 is not compatible with net7.0-windows10.0.19041 (.NETCoreApp,Version=v7.0) / win10-x64. Package Microsoft.Toolkit.Uwp.Connectivity 7.1.3 supports: uap10.0.17763 (UAP,Version=v10.0.17763)	efce3D_ble	D:\Development\efce3d-artemis\efce3D_ble_artemis\efce3D_ble\efce3D_ble.csproj	

I installed the package you sent.
Still, Uwp.Connectivity does not install (it shouldn'y anyway)

Though, WIndows Machine does not throw any more the NotImplemented errors!
I suppose that I initializing ble adapter in Windows Machine (the same code works fine in Android), but bluetooth status is returned always as ON (even if I had closed the Bluetooth on my Laptop), but do not crashes.
I executed the StartScanningForDevicesAsync, but it crashed with an error message (RPC_E_WRONG_THREAD) which is not known to me, though it doesn't look like a Plugin.BLE problem. I know that bluetooth stack should always run on the same thread and this might be a problem of my code. I just need a lot of time to debug!

janusw, 2023-01-08:
> @janusw thank you very much for your effort and prompt reply! Although you say that 3.0.0.beta.2 didn't support Maui .NET 7, I installed nuget package and my code worked for android.

Well, yes, I also could make beta.2 work with a net6.0-based app on Android and iOS. However, .NET 7 introduced several incompatibilities (see https://learn.microsoft.com/en-us/dotnet/core/compatibility/7.0), and in general is not compatible with .NET 6 (in particular on Windows, apparently).

> On WIndows Machine I faced the notImplement error and found that I need to install Microsoft.Toolkit.Uwp.Connectivity, which does not install as it is not compatible for versions ABOVE 10.0.17763: [...]

Ok, I think I see your problem now: When using Plugin.BLE with MAUI, you actually need `CommunityToolkit.WinUI.Connectivity` instead of `Microsoft.Toolkit.Uwp.Connectivity`. Sorry that I missed this earlier.

> Though, WIndows Machine does not throw any more the NotImplemented errors! I suppose that I initializing ble adapter in Windows Machine (the same code works fine in Android), but bluetooth status is [...]

Huh, that sounds like progress at least. Could you check whether adding `CommunityToolkit.WinUI.Connectivity` fixes this?

janusw, 2023-01-08:
> Though, WIndows Machine does not throw any more the NotImplemented errors! I suppose that I initializing ble adapter in Windows Machine (the same code works fine in Android), but bluetooth status is returned always as ON (even if I had closed the Bluetooth on my Laptop), but do not crashes.

Btw, this sounds like #642.

nefen, 2023-01-10:
@janusw , I checked again:
1. After installing the package you sent me, StartScanningForDevicesAsync() works in Windows Machine without problem (I was correct. RPC_E_WRONG_THREAD error had nothing to do with Plugin.BLE, but from my code. Android was not crashing but Windows Machine did. Anyhow I had placed software logic inside the bluetoothe service, whish is anyway a bad practice and I had corrected it).
2. CommunityToolkit.WinUI.Connectivity installs in Maui Net7., but till now, I cannot discover any difference! It works with or without it!

janusw, 2023-01-10:
> @janusw , I checked again:
>
>     1. After installing the package you sent me, StartScanningForDevicesAsync() works in Windows Machine without problem (I was correct. RPC_E_WRONG_THREAD error had [...]

Ok, thanks for checking.

>     2. CommunityToolkit.WinUI.Connectivity installs in Maui Net7., but till now, I cannot discover any difference! It works with or without it!

Yeah, that's expected since the connectivity toolkit will be pulled in as a secondary dependence via Plugin.BLE anyway.

closed 2025-03-26 by nefen