dotnet-bluetooth-le/dotnet-bluetooth-le #654: Enable Bluetooth and Location

onurcoskun14, 2023-02-13:
I am trying to enable Bluetooth and Location for my andriod application. I know how to request permissions but i couldn't enable them can anyone help ?

janusw, 2023-03-19:
> I am trying to enable Bluetooth and Location for my andriod application. I know how to request permissions but i couldn't enable them can anyone help ?

I'm not sure what exactly you mean.

Which Android API level are you targeting? Do you use Xamarin or MAUI?

onurcoskun14, 2023-03-20:
I use Xamarin and target Android API level is 13.
Basicly i want a button for opening Bluetooth and Location Service is they are not opened .

janusw, 2024-03-16:
Enabling/disabling Bluetooth on Android has recently been implemented via #823, see also issue #821. Does this fix this issue for you?