dotnet-bluetooth-le/dotnet-bluetooth-le #657: Filtering by service UUID is not working

davepruitt, 2023-03-09:
Hi,

I am currently having trouble filtering by service id when scanning. Here is my code for scanning:
```
var scan_filter_options = new ScanFilterOptions();
scan_filter_options.ServiceUuids = new [] { Guid.Parse(service_id) };
await ble_adapter.StartScanningForDevicesAsync(scan_filter_options);
```

This code returns _no devices_. I know that the device is properly advertising, because when I open nRF Connect on my phone, it is able to see the BLE device _and_ its advertised service.

Also, if I manually connect to the device and then get the service, it is returned to me as a valid service. So this code works:
```
private async void Handle_BLE_DeviceConnected(object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e)
{
    //Set the connected device
    ConnectedDevice = e.Device;

    //Get the required service from the PCM device
    if (ConnectedDevice != null)
    {
        var service = await ConnectedDevice.GetServiceAsync(Guid.Parse(service_id));
        
        //"service" is successfully obtained, and it is not null
        //I am able to successfully read/write characteristics        
    }
}
```

So manually connecting to the device and then getting its service _works_. But scanning and filtering by the same service UUID does _not work_.

Any suggestions?

## Steps to reproduce

See above explanation

## Expected behavior

Devices should be found when filtering by service UUID

## Actual behavior

No devices found when filtering by service UUID

### Crashlog

No crashes

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** Android 13 (.NET MAUI)

**Device:** Samsung Galaxy Tab A7 Lite

janusw, 2023-03-19:
Hm, hard to tell without further details. Some ideas:
* Are you sure that you are constructing the Guid correctly? What's the value of `service_id`? Note that [GuidExtension.UuidFromPartial](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b602ecebc2ffe0c23b830833501d7f738b105d18/Source/Plugin.BLE/Shared/Extensions/GuidExtension.cs#L12) can be useful here. (It's probably ok, since you say that you can obtain the service with the same ID after connecting. Just making sure ...)
* Some extensions of the scan filtering were implemented in version 2.1.3, e.g. #578 and #588. It's not impossible that one of these broke something. You could check version 2.1.2, in order to rule this out.
* What kind of advertisements do you receive from the device without filtering? How exactly do they look like (in particular, which record type)? You could try using a [ServiceDataFilter](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b602ecebc2ffe0c23b830833501d7f738b105d18/Source/Plugin.BLE/Shared/ScanFilterOptions.cs#L59), possibly with a UUID only and without actual data, in order to see if you get anything that way.

Hackmodford, 2023-05-26:
We ran into this issue and discovered that the ble device needs to report the service id in the advertisement record.

We had a device that had the expected service, but did not list it in the advertisement record. Once we added it there the scan filter worked correctly.

Also note, iOS was able to scan/filter the devices correctly without the record, while android would filter it out.

axa88, 2024-02-13:
@Hackmodford

>Also note, iOS was able to scan/filter the devices correctly without the record, while android would filter it out.

It seems implausible to be able to filter a service UUID from a scan which never added the data to the advertisement, it be very useful to know any detail if this is in fact the case.

Hackmodford, 2024-02-14:
@axa88 To be clear we noticed that iOS seemed to always report the correct device name, while Android seemed to cache an older device name even if the advertisement record was updated.

In order to fix this we used this extension.

```C#
/// <summary>
/// Returns the Complete Local Name from the advertisement record.
/// Falls back to the standard device name if the record is missing.
/// </summary>
/// <returns>Name of the device</returns>
public static string LatestName(this IDevice device)
{
    if (device.AdvertisementRecords != null)
    {
        foreach (var record in device.AdvertisementRecords)
        {
            if (record.Type != AdvertisementRecordType.CompleteLocalName)
            {
                continue;
            }

            if (record.Data.Length == 0)
            {
                continue;
            }
            return Encoding.UTF8.GetString(record.Data);
        }
    }
    return device.Name;
}
```