dotnet-bluetooth-le/dotnet-bluetooth-le #658: ScanFilterOptions.ManufacturerIds from the github repository description does not exist anymore

janseris, 2023-03-13:
## Steps to reproduce

1.  Create new MAUI .NET 7 App
2. paste scanFilterOptions.ManufacturerIds = new [] {1,2,3,etc}; // android only filter in source code when Plugin.BLE is installed

## Expected behavior
ScanFilterOptions.ManufacturerIds exists in Plugin.BLE library

## Actual behavior
Compile-time error with it not existing

## Configuration

Plugin.BLE 2.1.3

janseris, 2023-03-13:
> You haven't used our issue template
I have but deleted unnecessary parts

janusw, 2023-03-19:
You're right, `ManufacturerIds` was replaced by the more general `ManufacturerDataFilters` in 5cd2ce5c026b7d569ae5a742805a627b9b857e1f.

(Use the source, Luke: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/Plugin.BLE/Shared/ScanFilterOptions.cs)

So yes, the scan filtering section in the ReadMe needs an update.

janusw, 2023-03-20:
Fixed on master via 87d5e73f221f570c8bc7b88db09dd85f901b4fc6. Thanks for reporting!

closed 2023-03-20 by janusw