dotnet-bluetooth-le/dotnet-bluetooth-le #659: Add documentation to the NuGet package source code

janseris, 2023-03-14:
The Plugin.BLE NuGet package has no documentation for anything aavailable via Visual Studio.
This is important to be able to understand the behavior when using the library during development.
Please add documentation to the NuGet package source code.

Thank you

janusw, 2023-03-19:
Actually Plugin.BLE has a lot of documentation comments in the source code and AFAICS those are also visible in VS.

For example, if I hover over `CrossBluetoothLE.Current.BLE` in the VS code editor (in a project that uses Plugin.BLE via the nuget), then I see:
```
IAdapter IBluetoothLE.Adapter { get; }
Adapter that provides access to the physical Bluetooth adapter.
```

AFAICS this works well both in VS 2022 v17.5.2 (on Windows) and VS for Mac 2022 v17.5.2 (on MacOS), using the 3.0.0 beta nugets.

Are you missing documentation for a specific class or method? Or do you mean a different kind of documentation that what I'm pointing out here? Or with a different nuget or VS version? If so, please be more specific.

janseris, 2023-03-19:
@janusw 
Hello,
thank you for writing your exact setup. 
I updated to 17.5.2 today and had 2.2.0-pre5 installed until now.
I just updated to version 3,0.0 beta 2 and some documentation is now there and looks very understandable.

Thanks for providing this library.

closed 2023-03-19 by janseris