dotnet-bluetooth-le/dotnet-bluetooth-le #660: BLE devices are retuning device name NULL during scanning process on android 13

BasuHuddar, 2023-03-20:
We are not getting the BLE device name while scanning the Bluetooth devices in android 13.

## Steps to reproduce

1. Scan any BLE devices in android 13

## Expected behavior
We should get the device name in the scanned result

## Actual behavior
The device name is null

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** e.g. Android .. (including version!!! e.g. Android 13)

**Device:** e.g. Samsung, Redmi

janusw, 2023-03-20:
Similar observations have been reported before, e.g. in #456.

Is the device name missing for all devices that are returned by the scan, or only some of them?

BasuHuddar, 2023-03-21:
@janusw Yes, it is missing all the device's names in android 13.

sisaacks, 2023-03-29:
I am able to see the device name and device Id for some but  not for others.  HOWEVER since the latest Android update with Android 13, if a device drops connection, I cannot seed the device again unless I restart the app.  Then the device is seen an automatically connects again. 

when the device drops connection I have tried stopping and restarting the scanning but the device is never found again.  Other devices are, just not the device that lost connection.  I have tested this with a variety of products.  

I guess I should submit a bug for this as well

BasuHuddar, 2023-04-03:
@janusw, i got the fix for this, should we get access to the code to edit and generate a nupkg?

janusw, 2023-04-03:
@BasuHuddar If you have a fix, you could either just post a patch/snippet here or ideally open a Pull Request.

sisaacks, 2023-04-03:
> if (string.IsNullOrEmpty(BluetoothDevice.Name))
> {
> ByteBuffer order =
> ByteBuffer.Wrap(AdvertisementData).Order(ByteOrder.LittleEndian);
> while (order.Remaining() > 2)
> {
> byte b = (byte)order.Get();
> if (b != 0)
> {
> if (order.Get() != 9)
> {
> order.Position((order.Position() + b) - 1);
> }
> else
> {
> byte[] bArr2 = new byte[(b - 1)];
> order.Get(bArr2);
> try
> {
> Name = (System.Text.Encoding.UTF8.GetString(bArr2)).Trim();
> }
> catch (UnsupportedEncodingException unused)
> {
> Trace.Message("Error occured while converting byte to string",
> unused.Message);
> }
> }
> }
> }
> }
> else
> {
> Name = BluetoothDevice.Name;
> }
> @janusw <https://github.com/janusw> here is the code base we need to
> chage it device class in android

yeah that is not the same issue I am seeing.  I cannot see item if it drops
connection

janusw, 2023-04-04:
> @janusw here is the code base we need to chage it device class in android

Have you actually tried if this fixes the problem? I haven't looked at it very closely yet. Where exactly does this code need to go?

What you should do to go forward with this is:
* fork this repo and clone it (if you haven't done that)
* create a new branch, commit your changes and push it to your fork
* open a PR from the branch

The PR will actually generate a nupkg automatically. Then others can try and review it.

BasuHuddar, 2023-04-05:
@janusw  Yes it's fixed the problem for me, From which branch should i create the branch?

BasuHuddar, 2023-04-05:
[image]

Unable to build it getting errors, but I'm able to build tag 2.1.3, i tested here it's working
Same code i will add them to the master fork and i will create the pr to it.

@janusw are you ok with this?

BasuHuddar, 2023-04-05:
[image]

@janusw I tested in Tag 2.1.3 it working, PFA

janusw, 2023-04-06:
> Unable to build it getting errors, but I'm able to build tag 2.1.3, i tested here it's working Same code i will add them to the master fork and i will create the pr to it.
> 
> @janusw are you ok with this?

We will not have another 2.x release, so in the end it will need to be applied to master. If you have trouble building master, you can also start with a branch based on 2.1.3 and we forward-port it from there.

Or you port your code to master yourself. The CI will tell you if there are build errors.

BasuHuddar, 2023-04-06:
@janusw Here is my commit, check this and let me know if it's working or not.

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/d452eeaddcb5a984d8603025305b3c52775afab4

janusw, 2023-04-06:
> @janusw Here is my commit, check this and let me know if it's working or not.
> 
> [d452eea](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/d452eeaddcb5a984d8603025305b3c52775afab4)

Thanks for the commit. It seems you're taking the device name from advertisement records of type `CompleteLocalName`, right?

I think the idea as such is pretty good, but the technical implementation needs some adjustment IMHO:
* Instead of parsing the raw advertisment data yourself, you could use the `AdvertisementRecords` member of the `Device` class (which is generated by the method `ParseScanRecord`). Then you don't need to add a new member `AdvertisementData`.
* You could set the name based on the advertisement in the `Device` constructor already, and possibly update it via the `NativeDevice`'s name if that is non-empty (in the `Update` method).

BasuHuddar, 2023-04-07:
@janusw I dint get it, but i checked the code in the parseScanrecord you are using for getting the advertisement data, but how should we use it to assign the name of the BLE device?

One more before parsing the advertising byte we were setting the name in the update method.
If you know very well, could please apply this code there?

janusw, 2023-04-07:
> > @janusw I dint get it, but i checked the code in the parseScanrecord you are using for getting the advertisement data, but how should we use it to assign the name of the BLE device?
> 
> Let me take care of molding this into the right form. I'll also create a PR from it. Maybe you can test once more after I'm done ...

See #667.

janusw, 2023-04-09:
@BasuHuddar Two more questions here:
* Are you only seeing this with version 2.x, or also with 3.0 beta?
* On which host device do you observe the missing device names?

BasuHuddar, 2023-04-10:
@janusw Yes I'm getting the device name null in the 2.x and 3.0 beta version
Android 13(Samsung M33 5G)

janusw, 2023-04-17:
I recently noticed that in the  developer settings (networking section), there is an option to show Bluetooth devices without names (Mac addr only). @BasuHuddar Could you check if you have that switch enabled?

BasuHuddar, 2023-04-18:
@janusw I tried but no luck on that as well

janusw, 2023-04-18:
> @janusw I tried but no luck on that as well

What do you mean? You could not check it? Or it's not enabled?

If the developer mode is not enabled, try this:
https://developer.android.com/studio/debug/dev-options#enable

BasuHuddar, 2023-04-18:
> > @janusw I tried but no luck on that as well
> 
> What do you mean? You could not check it? Or "show devices without names" is not enabled?
> 
> If the developer mode is not enabled, try this: https://developer.android.com/studio/debug/dev-options#enable

I enable the developer mode and switched enabled for the Bluetooth devices without names (Mac address only), but it's not working

janusw, 2023-04-18:
> I enable the developer mode and switched enabled for the Bluetooth devices without names (Mac address only), but it's not working

Well, of course. If that switch is enabled, then you will certainly not see any device names.

The question is: Do you still fail to see device names if the switch is off?

Do I understand correctly that the device names are only missing on one particular device (Samsung M33), but not on others?

BasuHuddar, 2023-04-18:
> Well, of course. If that switch is enabled, then you will certainly not see any device names.
>
> The question is: Do you still fail to see device names if the switch is off?
>
> Do I understand correctly [...]

Yes, I haven't seen any device names, if it's switched off or switched on.

We are seeing this in Samsung M33 and Redmi phones(Android 13).

As I don't haves the other phone to check it here.

janusw, 2023-04-18:
> Yes, I haven't seen any device names, if it's switched off or switched on.

Ok, thanks for checking.

> We are seeing this in Samsung M33 and Redmi phones(Android 13).

AFAICS it is not a general Android 13 problem, since my Samsung S21+ seems to show device names properly.
 
> As I don't haves the other phone to check it here.

Hm, but in https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/667#issuecomment-1501558501 you said that you see CompleteLocalName ad records on a Samsung M40 on Android 11, right?

I still wonder how the patch that you proposed above can work if you don't see any CompleteLocalName records on the M33? This can only mean that the parsing of those records somehow fails in Plugin.BLE. It would be great if you could verify/debug this hypothesis, otherwise we'll not be able to make progress here, unfortunately (I cannot reproduce the issue after all).

janusw, 2023-04-20:
@BasuHuddar Could this possibly be related to #567? Do you see a `System.ArgumentException` during the parsing of the advertisement?

BasuHuddar, 2023-04-21:
> > > > I enable the developer mode and switched enabled for the Bluetooth devices without names (Mac address only), but it's not working
> > > 
> > > 
> > > Well, of course. If that switch is enabled, then you will certainly not see any device names.
> > > The question is: Do you still fail to see device names if the switch is off?
> > > Do I understand correctly that the device names are only missing on one particular device (Samsung M33), but not on others?
> > 
> > 
> > Yes, I haven't seen any device names, if it's switched off or switched on.
> 
> Ok, thanks for checking.
> 
> > We are seeing this in Samsung M33 and Redmi phones(Android 13).
> 
> AFAICS it is not a general Android 13 problem, since my Samsung S21+ seems to show device names properly.
> 
> > As I don't haves the other phone to check it here.
> 
> Hm, but in [#667 (comment)](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/667#issuecomment-1501558501) you said that you see CompleteLocalName ad records on a Samsung M40 on Android 11, right?
> Yes it's an Android 11
> I still wonder how the patch that you proposed above can work if you don't see any CompleteLocalName records on the M33? This can only mean that the parsing of those records somehow fails in Plugin.BLE. It would be great if you could verify/debug this hypothesis, otherwise we'll not be able to make progress here, unfortunately (I cannot reproduce the issue after all).
Yes debugged here, but I'm not finding it completeLocalname

BasuHuddar, 2023-04-21:
> @BasuHuddar Could this possibly be related to #567? Do you see a `System.ArgumentException` during the parsing of the advertisement?

**On My side I am not seeing System.ArgumentException during the parsing of the advertisement **

janusw, 2023-04-28:
Actually I found one condition, under which I can reproduce missing device names on Android (v13) ...

If I use a scan filter, which excludes `ShortLocalName` and `CompleteLocalName` records (e.g. a `ServiceDataFilter`), then I actually see no devices names for the discovered devices.
If I disable the scan filter, I observe `ShortLocalName` and `CompleteLocalName` records coming in and the devices names are set correctly.

@BasuHuddar Could your issue be related to scan filtering? Did you use a scan filter in your tests? Can you reproduce the issue when no scan filter is in effect?

paulobikiwelu, 2023-04-28:
@janusw 
I need help ... 😫 
Scanning for device does not return any device ...
I tried with Android Emulator 13 and also installed the  App on my phone (Samsung) but I could not see any device. 
I am using Nugget / version 2.1.3

paulobikiwelu, 2023-04-28:
private async void BluetoothAdapter_DeviceDiscovered(object sender, DeviceEventArgs e)
    {
        if (FindknownDevices(e.Device.Id) == null)
        {
            var deviceNode = new DeviceNode(e.Device);
            devices.Add(deviceNode);
            await Toaster.ShowAsync($"Found {e.Device.State.ToString().ToLower()} {e.Device.Name}.");
        }
    }

await bluetoothAdapter.StartScanningForDevicesAsync();

paulobikiwelu, 2023-04-28:
@janusw @BasuHuddar 

:) 😀 please help.

Can't see any device during Scanning...
I tried with Android Emulator 13 and also installed app on phone (Samsung) but was not successful on .net 6  MAUI

I am using API version 2.3.1

Thanks 😊 

private async void BluetoothAdapter_DeviceDiscovered(object sender, DeviceEventArgs e)
    {
        if (FindknownDevices(e.Device.Id) == null)
        {
            var deviceNode = new DeviceNode(e.Device);
            devices.Add(deviceNode);
            await Toaster.ShowAsync($"Found {e.Device.State.ToString().ToLower()} {e.Device.Name}.");
        }
    }

await bluetoothAdapter.StartScanningForDevicesAsync();

janusw, 2023-04-28:
> :) 😀 please help.
> 
> Can't see any device during Scanning... 

@paulobikiwelu Please open a new issue, this one is about missing device names.

paulobikiwelu, 2023-04-29:
> 
> > :) 😀 please help.
> > 
> > Can't see any device during Scanning... 
> 
> @paulobikiwelu Please open a new issue, this one is about missing device names.

@janusw 
Ok ..  I will do just that. 
Thanks 😊

vitalii-vov, 2024-05-18:
Maybe this will help some of you.
After some observation, it seems that the default scan mode is different on different android versions. On Android 14 it is `LowPower` and on Android 10 it is `Balanced`.
If you set this parameter to `Balanced` or `LowLatency` before starting the scan, the device names become visible.

```
public class BleScanner
{
    private readonly IAdapter _adapter;

    public BleScanner()
    {
        _adapter = CrossBluetoothLE.Current.Adapter;
        //  [FIX] Fixed android 14 scan device bug
        _adapter.ScanMode = ScanMode.Balanced;

        _adapter.ScanTimeout = 30 * 1000;
        _adapter.DeviceDiscovered += OnDeviceDiscovered;
        _adapter.ScanTimeoutElapsed += OnScanTimeoutElapsed;
    }

    public event EventHandler<IDevice> Discovered;

    public void Start()
    {
        StartScan();
    }

    public void Stop()
    {
        _adapter.StopScanningForDevicesAsync();
    }

    private void StartScan()
    {
        if (_adapter.IsScanning)
            return;

        //_adapter.ScanMode = ScanMode.Balanced;
        _adapter.StartScanningForDevicesAsync();
    }

    private void OnDeviceDiscovered(object sender, DeviceEventArgs e)
    {
        Discovered?.Invoke(this, e.Device);
    }

    private void OnScanTimeoutElapsed(object sender, EventArgs e)
    {
        //  auto restart scan
        StartScan();
    }
}
```

janusw, 2024-05-19:
> Maybe this will help some of you. After some observation, it seems that the default scan mode is different on different android versions. On Android 14 it is `LowPower` and on Android 10 it is [...]

Interesting observation. I wasn't aware of both of these points (that the default scan mode differs among Android versions, and that it influences the device names).

Actually, looking at the code, it seems that in Plugin.BLE the scan mode should always be set to `LowPower` by default ...

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/1e19fa858cded5eeb3c0ddd54a1aee05db14d4c1/Source/Plugin.BLE/Shared/AdapterBase.cs#L71-L76

@vitalii-vov Are you saying you observe something different? Or was your statement a general one, independent of Plugin.BLE?

vitalii-vov, 2024-05-19:
Just double-checked on Android 10 and Android 14. The default value is `LowPover`. I was indeed mistaken.

janusw, 2024-05-20:
> Just double-checked on Android 10 and Android 14. The default value is `LowPover`.

Ok, thanks for checking.

If the ScanMode really has an influence on the presence of device names, I guess one could think about setting the default to `Balanced`, so that the names are present by default.

However, that sort of effect would be kind of surprising to me. @vitalii-vov Could you please double-check if you really observe a difference between `LowPower` and `Balanced`?

janusw, 2024-05-20:
> If the ScanMode really has an influence on the presence of device names, I guess one could think about setting the default to `Balanced`, so that the names are present by default.
>
> However, that [...]

Actually I just tried it on a Samsung Galaxy S21+ (Android 14) with the BLE.Client.Maui sample app, and I can not confirm your observation. I see devices with names also when using LowPower mode (they have ShortLocalName or CompleteLocalName advertisements).

vitalii-vov, 2024-06-08:
I apologize, my comment was in error. Changing the scan mode does not solve this problem for me. The reason was somehow related to the fact that I had two bluetooth libraries in my project (the second one is shiny), and the second library was somehow affecting the behavior of this library. So far I can't say anything important on the merits.

vitalii-vov, 2024-06-08:
I apologize, my comment was in error. Changing the scan mode does not solve this problem for me. The reason was somehow related to the fact that I had two bluetooth libraries in my project (the second one is shiny), and the second library was somehow affecting the behavior of this library. So far I can't say anything important on the merits.

I can only say that this problem is not present in shiny on MAUI. I tested shiny, Plugin.BLE and 32feet. The only library where this problem is solved is shiny, but I can't use it on MAUI because it doesn't support Windows.