dotnet-bluetooth-le/dotnet-bluetooth-le #661: WriteAsync throws System Exception Status AccessDenied

Troltsch, 2023-03-20:
## Steps to reproduce

1. Scan for bluetooth devices

2. Connect to specific bluetooth device

3. Get Services & Characteristics

4. Write to specific characteristic:
`if(MainThread.IsMainThread) await characteristic.WriteAsync(data);`

## Expected behavior
Writes data to connected device without any exception.
Reading values from characteristics doesn't result in any exceptions and works as intended to.

## Actual behavior
An System Exception is thrown with status: AccessDenied (see Crashlog).

Bluetooth Capabilities are checked.

### Crashlog

System.Exception: [WriteNativeAsync] failed with status: AccessDenied
   at Plugin.BLE.Extensions.GattResultExtensions.ThrowIfError(GattCommunicationStatus status, String tag, Nullable`1 protocolError)
   at Plugin.BLE.Extensions.GattResultExtensions.ThrowIfError(GattWriteResult result, String tag)
   at Plugin.BLE.UWP.Characteristic.WriteNativeAsync(Byte[] data, CharacteristicWriteType writeType)
   at Plugin.BLE.Abstractions.CharacteristicBase`1.WriteAsync(Byte[] data, CancellationToken cancellationToken)
   at MatrixMAUI.BluetoothManager.Write(ICharacteristic characteristic, Byte[] data)

## Configuration

**Version of the Plugin:* 3.0.0-beta.2

**Platform:** Windows MAUI net7.0-windows10.0.19041.0

CliffAgius, 2024-09-27:
Ok so looking into this it seems the method used for writing the data is from UWP and not WinUI which is what MAUI targets.  Looking to see if there is a replacement method in WinUI and it seems there isn't see [See Here!](https://learn.microsoft.com/en-us/windows/communitytoolkit/helpers/bluetoothlehelper#methods-2)

So looks like Windows is blocked for MAUI until they add a way to write at least that is my quick and dirty look into this error.

janusw, 2024-10-10:
> it's a pain as I needed the Windows version to work with the Droid and iOS.

I don't understand this sentence 🤔 

> Ok so looking into this it seems the method used for writing the data is from UWP and not WinUI which is what MAUI targets. Looking to see if there is a replacement method in WinUI and it seems [...]

Which "method used for writing" do you actually mean? This one?
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/09b1b2c4dafdc26f4491caa11a99b44c4d98b461/Source/Plugin.BLE/Windows/Characteristic.cs#L84-L92

AFAIK, nothing in the latest version of Plugin.BLE is UWP- or WinUI-specific any more. The Windows implementation should be completely generic. Right, @AskBojesen ?

SysadmEgooHealth, 2024-10-10:
> I don't understand this sentence 🤔
>
>
> Which "method used for writing" do you actually mean? This one? [...]

YES! The WIndows implementation refers ONLY: Windows.Devices.Bluetooth.*  and Windows.Devices.Enumeration namespaces and no additional packages.

What might be challenging to users are the Windows supported frameworks:
uap10.0.19041;net7.0-windows10.0.19041;net8.0-windows10.0.19041;net8.0-windows10.0.22000.0

Which framework are you targeting @CliffAgius ?

AskBojesen, 2024-10-10:
> I don't understand this sentence 🤔
>
>
> Which "method used for writing" do you actually mean? This one? [...]

YES! The WIndows implementation refers ONLY: Windows.Devices.Bluetooth.*  and Windows.Devices.Enumeration namespaces and no additional packages.

What might be challenging to users are the Windows supported frameworks:
uap10.0.19041;net7.0-windows10.0.19041;net8.0-windows10.0.19041;net8.0-windows10.0.22000.0

**Which framework are you targeting @CliffAgius / @Troltsch ?**