dotnet-bluetooth-le/dotnet-bluetooth-le #662: With .Net7 supporting coming to the beta, is there a plan to release .Net6 out of beta?

IainS1986, 2023-03-21:
I saw a comment about PRS coming with .Net7 support into the next 3.0.0 beta (maybe already in there?)

Is there a plan to get a 3.0.0 full release for at least .Net6?

janusw, 2023-03-21:
Yes, I think we should have a final version 3.0.0 with both .net6 and .net7 support soon.

IainS1986, 2023-04-17:
Awesome news. We're still on Xamarin Native but I'm looking to finally pull the plug and do the .Net6/7 update in the coming months. Might just try it out at some point to see how painful the process is 

What sort of API changes are coming in 3.0? There any notes anywhere currently?

janusw, 2023-04-18:
> Awesome news. We're still on Xamarin Native but I'm looking to finally pull the plug and do the .Net6/7 update in the coming months.

I don't think you're alone here. Probably many people have not finished the transition yet. From my point of view, MAUI is not yet where it should be (quality-wise).

Btw, v3.0 will also continue to support Xamarin for a while, so you might consider trying out 3.0.0-beta.3 with your Xamarin app.

> Might just try it out at some point to see how painful the process is

It depends. For small apps it might be rather straightforward, but for large and complex projects, the migration can certainly be quite involved. Unfortunately I don't know any good tools that automate the process, but at least there are a number of useful upgrade guides on the web.

For Xamarin Native, the migration is probably a bit more painful than for Xamarin.Forms (but I'm not sure if it makes much difference).

> What sort of API changes are coming in 3.0? There any notes anywhere currently?

In particular #668/#669 will be coming up soon, but I hope we'll not have many breaking changes beyond that.

IainS1986, 2023-04-18:
Cheers for the heads up - status code handling doesn't seem too much of a change to update 👍 

Personally, we are avoiding MAUI completely - we're just going to upgrade Xamarin Native ->  .net iOS and .net Android and continue 'native' development (MvvmCross based app).

Going Xamarin Native -> Forms/MAUI just does not seem remotely worth the effort (and think we'd be worse off to be frank).

janusw, 2023-04-18:
> Personally, we are avoiding MAUI completely - we're just going to upgrade Xamarin Native -> .net iOS and .net Android and continue 'native' development (MvvmCross based app).

Ah, I get it. In that case, the upgrade to .NET 6 should be much easier than going from Xamarin.Forms to MAUI, I assume.

> Going Xamarin Native -> Forms/MAUI just does not seem remotely worth the effort (and think we'd be worse off to be frank).

Depends on the use case, I guess. Having cross-platform UI in a single code base is a great thing in principle. Both Xamarin.Forms and MAUI certainly have their pain points, though.

I will close this issue now (it rather belong into https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/discussions anyway) ...

closed 2023-04-18 by janusw