dotnet-bluetooth-le/dotnet-bluetooth-le #664: Could not load assembly Plugin.BLE.Abstractions

DanielAntos90, 2023-03-28:
## Steps to reproduce

Use 3.0.0-beta.2 on [MauiXAMLBluetoothLE]( https://github.com/TimoKinnunen/MauiXAMLBluetoothLE)
You can check changes in forked repo [here]( https://github.com/DanielAntos90/MauiXAMLBluetoothLE)

## Expected behavior

## Actual behavior

on Android API 31 I am getting this error in file Xamarin.Android.Common.targets line 1438:

System.IO.FileNotFoundException: Could not load assembly 'Plugin.BLE.Abstractions, Version=1.0.0.0, Culture=neutral, PublicKeyToken='. Perhaps it doesn't exist in the Mono for Android profile?

I works fine when I switch version back to 2.2.0-prev5

## Configuration

**Version of the Plugin: 3.0.0-beta.2

**Platform:  Android 12.0 API 31

**Device: Xiaomi Redmi Note 11

janusw, 2023-04-05:
Does this also occur with 3.0.0-beta.1?

DanielAntos90, 2023-04-11:
I finally manage to run it using 3.0.0-beta.2, but I do not exactly know that the issue was.

nowlights, 2023-04-12:
i try using with 2.2.0-pre5, 3.0.0-beta.1 and beta.2 and i have the same error: 
`System.TypeLoadException: Could not resolve type with token 01000051 from typeref (expected class 'Plugin.BLE.Abstractions.Contracts.IBluetoothLE' in assembly 'Plugin.BLE, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null')`

i using Xamarin with netstandard2.1
Xamarin.Android version 13.2

DJVova, 2023-04-18:
> ## Steps to reproduce
> Use 3.0.0-beta.2 on [MauiXAMLBluetoothLE](https://github.com/TimoKinnunen/MauiXAMLBluetoothLE) You can check changes in forked repo [...]

This error appearing after upgrade Plugin.BLE to version 3.0.0* and disappearing after rebuilding solution.

janusw, 2023-04-18:
> This error appearing after upgrade Plugin.BLE to version 3.0.0* and disappearing after rebuilding solution. You need only rebuild your solutions and problem will disappear. We can close this issue.

Thanks for reporting this. I'll close the issue.

closed 2023-04-18 by janusw