dotnet-bluetooth-le/dotnet-bluetooth-le #668: [Feature Request] Return Error Codes on Read/Write

chrimaka, 2023-04-09:
On Read/Write operations there can be other error codes provided by the device giving a hint to the application what's going on (and what the actual problem is). Currently these error code are ignored. The intent is to pass the error codes as a result of a read or write operation.

chrimaka, 2023-04-09:
I'd like to add

```
default:
    exception = new Exception($"GattStatus: {(int)status}");
    break;
```

to Android's `GattCallback.cs`. 

From a design perspective - how should we pass the error back the application? We could pass the error code as a result of the read (or write) operation. This would change the API from `protected abstract Task<byte[]> ReadNativeAsync();` to  `protected abstract Task<Tuple<int, byte[]>> ReadNativeAsync();`. (with the major change to 3.x this would be okay from my perspective). The other option is, the throw a dedicate exception. The question is, if a regular error should result in an exception. 

@janusw do you have a though here?

janusw, 2023-04-09:
A PR would be very welcome 😉. We haven't released 3.0 yet, so an API change would be acceptable at this stage. Give me a moment to look into the details here.

janusw, 2023-05-18:
This is fixed via #669.