dotnet-bluetooth-le/dotnet-bluetooth-le #672: [Feature Request] Obtain Versionumber and Build Variant (Debug/Release) at Runtime

chrimaka, 2023-04-11:
The idea would be a method, that allows you to get the version of the current lib + the build variant at run time. this can be helpful to figure out what kind of apps are in the field.

janusw, 2023-04-16:
Hm, couldn't you simply determine the version from the nupkg / assembly metadata?

Something like:

```
var version_str = Assembly.GetAssembly(typeof(CrossBluetoothLE)).GetName().Version.ToString();
```

(see e.g. https://stackoverflow.com/questions/34775712/how-can-i-programatically-get-the-version-of-a-nuget-package-being-used-in-the-s)

janusw, 2023-04-16:
> Hm, couldn't you simply determine the version from the nupkg / assembly metadata?
> 
> Something like:
> 
> ```
> var version_str = Assembly.GetAssembly(typeof(CrossBluetoothLE)).GetName().Version.ToString();
> ```

However, it seems like this currently returns `1.0.0.0` (because that is what is set for the AssemblyVersion in the csproj file). Can (and should) be fixed, of course (but not for past releases, obviously) ...

janusw, 2023-04-16:
> However, it seems like this currently returns `1.0.0.0` (because that is what is set for the AssemblyVersion in the csproj file). Can (and should) be fixed, of course (but not for past releases, [...]

This should be fixed by 278171d214b9a9b0082d2c033bae39365c491216. But: The version that one gets from the assembly now is `3.0.0.0` instead of `3.0.0-beta.2` (because the pre-release label is dropped).