dotnet-bluetooth-le/dotnet-bluetooth-le #673: GattCallback Error 238 or 133 on Android

patriciaOrtuno28, 2023-04-12:
## Steps to reproduce

I am trying to connect to a BLE device, using the following code for scanning and then connection:

````csharp
            Device.BeginInvokeOnMainThread(async () =>
            {
                try
                {
                    TurnOnActivityIndicator();
                    adapter.ScanMode = ScanMode.LowPower;
                    devices.Clear();
                    var pairedDevices = adapter.GetSystemConnectedOrPairedDevices();
                    foreach (var dev in pairedDevices) { devices.Add(dev); }

                    var deviceNames = new List<string>();
                    adapter.DeviceDiscovered += (s, a) =>
                    {
                        if (!devices.Any(d => d.Id == a.Device.Id)) // Check if the device name is already in the list
                        {
                            devices.Add(a.Device);
                        }
                    };
                    await adapter.StartScanningForDevicesAsync();

                    foreach (var device in devices)
                    {
                        deviceNames.Add(device.Name);
                    }
                    TurnOffActivityIndicator();

                    // Display the list of device names
                    var selectedName = await DisplayActionSheet(
                        "Select a device",
                        "Cancel", null,
                        deviceNames.ToArray());

                    if (selectedName != "Cancel")
                    {
                        await adapter.StopScanningForDevicesAsync();
                        // Get the selected device
                        foreach (var device in devices)
                        {
                            if (device.Name == selectedName)
                            {
                                selectedDevice = device;
                                break;
                            }
                        }

                        // Connect to the selected device
                        try
                        {
                            var parameters = new ConnectParameters(forceBleTransport: true);
                            //await adapter.ConnectToKnownDeviceAsync(selectedDevice.Id, parameters);
                            await adapter.ConnectToDeviceAsync(selectedDevice);
                        }
                        catch (Exception ex) { Console.WriteLine(ex.ToString()); }

                        Console.WriteLine(selectedDevice.State);

                        if (selectedDevice.State == DeviceState.Connected)
                        {
(...)
````

## Expected behavior

The desired behavior would be that when the function that holds this code is invoked, it will show a list of all bonded and newly scanned devices and, when one is selected, it stops scanning and tries to connect to it. Then it will get its characteristics, etc.

## Actual behavior
On an iOS device it works without an issue, but, when using Android devices, it won't work.
If i use the following line of code:
````csharp
await adapter.ConnectToDeviceAsync(selectedDevice);
````
Then i get the exception shown in the Crashlog section.

However, if I use these lines:
````csharp
var parameters = new ConnectParameters(forceBleTransport: true);
await adapter.ConnectToDeviceAsync(selectedDevice, parameters);
````
It shows that it has indeed connected, but the reality is that it hasn't. The BLE device i am using hasn't changed the led color indicating the connection and, after trying to send information to the phone, it doesn't receive it. It has not connected even if it shows it has.

### Crashlog

The output of the execution of this code until reaching an Exception when using the `ConnectToDeviceAsync()` is the following when using Android 12 on a Xiaomi Redmi Note 11:
````
[BluetoothAdapter] isLeEnabled(): ON
[BluetoothLeScanner] onScannerRegistered() - status=0 scannerId=9 mScannerId=0
[BluetoothAdapter] isLeEnabled(): ON
[IS_CTS_MODE] true
[MULTI_WINDOW_SWITCH_ENABLED] false
[DecorView[]] onWindowFocusChanged hasWindowFocus false
[DecorView[]] onWindowFocusChanged hasWindowFocus true
[BluetoothGatt] connect() - device: 60:09:C3:E0:4D:D4, auto: false, eattSupport: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=550c57df-dd4b-4292-ba33-250cb7a6d35a
[BluetoothGatt] onClientRegistered() - status=0 clientIf=9
[OpenGLRenderer] endAllActiveAnimators on 0xb4000076781d3900 (AlertController$RecycleListView) with handle 0xb4000076781e0f20
[DecorView[]] onWindowFocusChanged hasWindowFocus true
[BluetoothGatt] onClientConnectionState() - status=238 clientIf=9 device=60:09:C3:E0:4D:D4
Thread started:  #22
[BluetoothGatt] close()
[BluetoothGatt] unregisterApp() - mClientIf=9
Plugin.BLE.Abstractions.Exceptions.DeviceConnectionException: GattCallback error: 238
  at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token) [0x000d0] in /Users/weija001/github/xamarin-bluetooth-le/Source/Plugin.BLE.Abstractions/Utils/TaskBuilder.cs:49 
  at Plugin.BLE.Abstractions.AdapterBase.ConnectToDeviceAsync (Plugin.BLE.Abstractions.Contracts.IDevice device, Plugin.BLE.Abstractions.ConnectParameters connectParameters, System.Threading.CancellationToken cancellationToken) [0x0014e] in /Users/weija001/github/xamarin-bluetooth-le/Source/Plugin.BLE.Abstractions/AdapterBase.cs:113 
  at Lockgene.Views.ProfilePage.<bt_button>b__28_0 () [0x00473] in (...)
Disconnected
[IS_CTS_MODE] true
[MULTI_WINDOW_SWITCH_ENABLED] false
````

The crashlog is exactly the same when using a Samsung Galaxy S10+ on Android 12, but the *GattCallback error is 133 instead of 238*.

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** 
- Android 12
- Android 12

**Device:** 
- Xiaomi Redmi Note 11
- Samsung Galaxy S10+

Haukinger, 2024-05-02:
Any news on this? I get GattCallback Error 133 when trying to connect to an Arduino Nano 33

acaliaro, 2025-01-23:
any news on this? 133 here