dotnet-bluetooth-le/dotnet-bluetooth-le #674: iOS Throws exception on startup in Maui. "This functionality is not implemented in the portable version of this assembly."

orbian, 2023-04-17:
I am trying to get my app running on iOS using Plugin.BLE 2.1.3.  It works on Android devices but when I deploy to my iPhone I get an exception as soon as I try to access Plugin.BLE.CrossBluetoothLE.Current in my constructor.
 
The exception message is "2023-04-17 08:01:01.008 Xamarin.PreBuilt.iOS[454:7612] This functionality is not implemented in the portable version of this assembly.  You should reference the NuGet package from your main application project in order to reference the platform-specific implementation."

## Steps to reproduce

1.
Create a .Net Maui app in VS 2022

2.
Add NuGet package Plugin.BLE 2.1.3

3.
Access something in the package like Plugin.BLE.CrossBluetoothLE.Current (which I assume is a function call behind the scenes).

## Expected behavior
The getter should return the value.

## Actual behavior
The app throws an exception with message:
"2023-04-17 08:01:01.008 Xamarin.PreBuilt.iOS[454:7612] This functionality is not implemented in the portable version of this assembly.  You should reference the NuGet package from your main application project in order to reference the platform-specific implementation."

### Crashlog

This is the stack when the exception is thrown:
>	0x8A in MauiApp1.MainPage..ctor at C:\Users\Bob\Source\Repos\MauiApp1\MauiApp1\MainPage.xaml.cs:50,13	C#
 	[External Code]	
 	0xD in MauiApp1.Program.Main at C:\Users\Bob\Source\Repos\MauiApp1\MauiApp1\Platforms\iOS\Program.cs:13,3	C#

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** iOS 16.4.1

**Device:** iPhone 11

janusw, 2023-04-17:
> I am trying to get my app running on iOS using Plugin.BLE 2.1.3. It works on Android devices but when I deploy to my iPhone I get an exception as soon as I try to access [...]

Version 2.1.3 does not support MAUI. Please use a 3.0 beta release. `3.0.0-beta.3` was released today.

closed 2023-04-17 by janusw

orbian, 2023-04-17:
Thanks. It's working now.