dotnet-bluetooth-le/dotnet-bluetooth-le #677: Advertising interval

midoragh, 2023-04-23:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

I like to show the advertising interval (and rssi but this is an other issue) of a device continuously during a scan and propose to add an additional optional parameter to StartScanningForDevicesAsync (i.e. bool reportAlways = false). If true OnScanDeviceDiscovered should be called always for the device and not just for the first advertising of the device.

I'm using OnDeviceAdvertised as a workaround to calculate the duration between calls for the same device (using a low-pass filter) but it would be easier to handle it in OnScanDeviceDiscovered. 

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** Android 10, API 29

**Device:** Wiko

janusw, 2023-04-24:
> I like to show the advertising interval (and rssi but this is an other issue) of a device continuously during a scan

AFAICS both should be possible already, right?

> I'm using OnDeviceAdvertised as a workaround to calculate the duration between calls for the same device (using a low-pass filter) but it would be easier to handle it in OnScanDeviceDiscovered.

Maybe I'm misunderstanding your request, but actually it sounds like the `DeviceAdvertised` event is more suitable for your purpose (continuously monitoring RSSI and advertising interval). Why do you want to use `DeviceDiscovered`?

> and propose to add an additional optional parameter to StartScanningForDevicesAsync (i.e. bool reportAlways = false). If true OnScanDeviceDiscovered should be called always for the device and not just for the first advertising of the device.

Note that `StartScanningForDevicesAsync` already has a parameter called `allowDuplicatesKey` that seems somewhat related (but only has an effect on iOS). In any case, you should probably use `DeviceAdvertised` if you want more that just the first advertising message ...

midoragh, 2023-04-25:
You are right. I checked HandleDiscoveredDevice how DeviceAdvertised and DeviceDiscovered are called.
So I withdraw the equest to enhance StartScanningForDevicesAsync.
Looking to HandleDiscoveredDevice I'm still wondering, why I'm not getting an rssi value in DeviceDiscovered if the the device is bonded. It works in DeviceAdvertised for the bonded and not bonded case. So why not some source lines later? ;-)

janusw, 2023-04-25:
> Looking to HandleDiscoveredDevice I'm still wondering, why I'm not getting an rssi value in DeviceDiscovered if the the device is bonded. It works in DeviceAdvertised for the bonded and not bonded case. So why not some source lines later? ;-)

Maybe because in that case the device is "discovered" simple because it's bonded (and without seeing an actual advertisement, so that there is no RSSI)?

midoragh, 2023-04-25:
There are a lot of bonded devices on the mobile, but only the device that advertises will be discovered on scanning.
And why does DeviceAdvertised reports it but not DeviceDiscovered some lines below. Maybe some thread sets it to zero.

I also have seen that DeviceAdvertised sometimes only provides AdvertisedRecords on the second call to DeviceAdvertised for this device.  That's not a big issue in the app but maybe a hint.