dotnet-bluetooth-le/dotnet-bluetooth-le #682: Unable to Find BLuetooth Devices after Scanning

paulobikiwelu, 2023-04-29:
Can't see any device during Scanning...
I tried with Android Emulator 13 and also installed app on phone (Samsung) but was not successful on .net 6 MAUI

I am using API version 2.3.1

```
public partial class DeviceScanner
{
[Inject]
public BluetoothService BluetoothService { get; set; }

private Status workflow;
public string workflowText;

private HashSet<DeviceNode> devices = new();
private DeviceNode selectedDevice = null;
private SemaphoreSlim semaphoreSlim = new(1, 1);

private async void OnSearch()
{
    await semaphoreSlim.WaitAsync();
    try
    {
        devices.Clear();
        selectedDevice = null;

        SetWorkflow("Scanning for Devices...", Status.Lock);
        await Task.Delay(1500);

        devices = await BluetoothService.CrossBluetooth.StartDiscoveryAsync(Callback);

        if (devices.Count == 0)
        {
            await Toaster.ShowAsync("Unable to find nearby Bluetooth LE devices. Try again.");
        }
    }
    finally
    {
        var successText = "Scan completed. Select a device and Connect";
        var failureText = "Unable to find nearby Bluetooth LE devices. Try again.";
        var text = devices.Count > 0 ? successText : failureText;
        UnlockWorkflow(text);
    }

}
```

 *standalone git repository** 

## Expected behavior
I expected to recieve visible BLE devices after Scanning 

## Actual behavior
After calling  "devices = await BluetoothService.CrossBluetooth.StartDiscoveryAsync(Callback);" i do not recieve any device , no Bluetooth Device could be seen.

## Configuration

	<application android:allowBackup="true" android:icon="@mipmap/appicon" android:roundIcon="@mipmap/appicon_round" android:supportsRtl="true"></application>
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.INTERNET" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
	<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />

**Version of the Plugin . 2.3.1

**Platform:  Android Emulator 13 >

**Device:** e.g. HTC Sensation /i Phone 7 ...

paulobikiwelu, 2023-04-29:
@janusw

janusw, 2023-04-29:
> After calling "devices = await BluetoothService.CrossBluetooth.StartDiscoveryAsync(Callback);" i do not recieve any device , no Bluetooth Device could be seen.

It's totally unclear what you're doing. There is no `StartDiscoveryAsync` method in the API of this library (Plugin.BLE). Are you using some other library maybe?

> _standalone git repository_* [Repo](https://github.com/bonifacebassey/Fruit.Electronic.SmartApp.git)

Also: This link is dead.

bonifacebassey, 2023-04-29:
Hi Janusw,

I followed these step to search for bluetooth devices using the  (Plugin.BLE) library but was not able to discover any device in range

1. Initialize and register callback

`var bluetoothLE = CrossBluetoothLE.Current;
var adapter = CrossBluetoothLE.Current.Adapter;
adapter.ScanTimeout = 5000;
adapter.ScanMode = ScanMode.LowLatency;

adapter.DeviceDiscovered += OnDeviceDiscovered;

private void OnDeviceDiscovered(object sender, DeviceEventArgs e)
{
        Console.WriteLine($"Found {e.Device.State.ToString().ToLower()} {e.Device.Name}.");
}`

2. Check and granted the following permissions

`(Android.Manifest.Permission.Bluetooth, true)
(Android.Manifest.Permission.BluetoothAdmin, true)
(Android.Manifest.Permission.BluetoothScan, true)
(Android.Manifest.Permission.BluetoothConnect, true)
(Android.Manifest.Permission.AccessFineLocation, true)
(Android.Manifest.Permission.AccessCoarseLocation, true)
(Android.Manifest.Permission.AccessBackgroundLocation, true)`

3. Start scanning for devices

`await adapter.StartScanningForDevicesAsync();`

Unfortunately, I wasn't able to discover any devices

janusw, 2023-04-30:
> I followed these steps to search for bluetooth devices using the (Plugin.BLE) library but was not able to discover any device in range
> 
>     1. Initialize and register callback
>     2. Check and granted the following permissions
>     3. Start scanning for devices
> 
> Unfortunately, I wasn't able to discover any devices

In addition to using the permissions in the Manifest, you will also need to request them at runtime, see:
* https://developer.android.com/guide/topics/connectivity/bluetooth/permissions
* https://developer.android.com/training/permissions/requesting

Starting with .NET 8, MAUI will have full support for the Bluetooth permissions (see https://github.com/dotnet/maui/pull/12413). Until then, you can do something similar to what we do in the sample client (which is Xamarin-based), see
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/BLE.Client/BLE.Client.Droid/Helpers/PlatformHelpers.cs

bonifacebassey, 2023-04-30:
> > I followed these steps to search for bluetooth devices using the (Plugin.BLE) library but was not able to discover any device in range
> > ```
> > 1. Initialize and register callback
> > 2. Check and granted the following permissions
> > 3. Start scanning for devices
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Unfortunately, I wasn't able to discover any devices
> 
> In addition to using the permissions in the Manifest, you will also need to request them at runtime, see:
> 
> * https://developer.android.com/guide/topics/connectivity/bluetooth/permissions
> * https://developer.android.com/training/permissions/requesting
> 
> Starting with .NET 8, MAUI will have full support for the Bluetooth permissions (see [dotnet/maui#12413](https://github.com/dotnet/maui/pull/12413)). Until then, you can do something similar to what we do in the sample client (which is Xamarin-based), see https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/BLE.Client/BLE.Client.Droid/Helpers/PlatformHelpers.cs

Yes i also requested permission at runtime.

janusw, 2023-04-30:
> Yes i also requested permission at runtime.

Well, then you're most probably doing something wrong somewhere. Hard to tell without a code sample for reproduction (or a working crystal ball) ...

janusw, 2023-04-30:
Coming back to the original bug report ...

> Can't see any device during Scanning... I tried with Android Emulator 13 and also installed app on phone (Samsung) but was not successful on .net 6 MAUI
>
> **Version of the Plugin . 2.3.1

I assume you mean version 2.1.3 here? Note that for MAUI (with .NET 6 or 7) you should use 3.0.0-beta.3.

bonifacebassey, 2023-04-30:
> > Yes i also requested permission at runtime.
> 
> Well, then you're most probably doing something wrong somewhere. Hard to tell without a code sample for reproduction (or a working crystal ball) ...

I will try it out again in some days and share a Repo to the source code if it doesn't work

volcano619, 2023-06-26:
@janusw 

Hi, I'm having the same issue, I'm unable to scan devices in android, but in iOS it is working fine, I have installed 3.0.0-beta4

Sharing my code

`using System;
using System.Collections.ObjectModel;
using System.Text;
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using Plugin.BLE.Abstractions.EventArgs;
using Plugin.BLE.Abstractions.Extensions;
using Xamarin.Forms;

namespace BluetoothPrinterExample
{
    public partial class MainPage : ContentPage
    {
        private IBluetoothLE ble;
        private IAdapter adapter;
        private IDevice selectedDevice;
        public ObservableCollection<IDevice> DeviceList = new ObservableCollection<IDevice>();

        public MainPage()
        {
            InitializeComponent();

            ble = CrossBluetoothLE.Current;
            adapter = ble.Adapter;
            adapter.DeviceDiscovered += DeviceDiscovered;
            DeviceList = new ObservableCollection<IDevice>();
        }

        private async void ScanButton_Clicked(object sender, EventArgs e)
        {
            DeviceList = new ObservableCollection<IDevice>();
            if (!ble.IsAvailable || !ble.IsOn)
            {
                await DisplayAlert("Error", "Bluetooth is not available or enabled.", "OK");
                return;
            }

            DeviceListView.ItemsSource = DeviceList;
            //adapter.ScanTimeout = 10000; // Scan for 10 seconds
            //adapter.ScanMode = ScanMode.Balanced;
            await adapter.StartScanningForDevicesAsync();
        }

        private async void DeviceDiscovered(object sender, DeviceEventArgs e)
        {
            DeviceList.Add(e.Device);
            // Filter and select your label printer based on device name, service UUID, or other criteria
            if (e.Device.Name != null && e.Device.Name.Contains("MATT"))
            {
                await adapter.StopScanningForDevicesAsync();
                selectedDevice = e.Device;
                await DisplayAlert("Success", "Printer found: " + selectedDevice.Name, "OK");
            }
        }

        private async void PrintButton_Clicked(object sender, EventArgs e)
        {
            if (selectedDevice == null)
            {
                await DisplayAlert("Error", "Please select a printer first.", "OK");
                return;
            }

            try
[... 68 lines cut ...]
            Xamarin.Essentials.Platform.OnRequestPermissionsResult(requestCode, permissions, grantResults);

            base.OnRequestPermissionsResult(requestCode, permissions, grantResults);
        }
    }

}`

I've used your platform helper, instead of a dependency service I've used as a static class and called it in the main activity itself, should be fine I suppose

`<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="1" android:versionName="1.0" package="com.test.bluetoothprinterexample">
	<uses-sdk android:minSdkVersion="21" android:targetSdkVersion="33" />
	
	<application android:label="BluetoothPrinterExample.Android" android:theme="@style/MainTheme"></application>
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
	<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
	<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
</manifest>`

volcano619, 2023-06-26:
the DeviceDiscovered even is not getting triggered

robotdoly, 2023-06-27:
@volcano619 could you find a solution?

janusw, 2023-07-03:
> @volcano619 could you find a solution?

Sounds like his problem was solved by this:

> had to add Manifest.Permission.AccessCoarseLocation, Manifest.Permission.AccessFineLocation in the runtime permissions

I'm closing this issue. As soon as you do it right, it suddenly starts working ;)

closed 2023-07-03 by janusw