dotnet-bluetooth-le/dotnet-bluetooth-le #686: Provide IsConnectable in advertised/discovered device

midoragh, 2023-05-06:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

None

## Expected behavior
The device should provide IsConnectable flag. This bool is available on Android in ScanResult and on iOS in CBAdvertisement.

## Actual behavior
The flag is not available

### Crashlog

no crash

## Configuration

**Version of the Plugin:** 1.2.3

**Platform:** Android 10, Api 29

**Device:** Wiko

midoragh, 2023-05-06:
I would prefer this device solution instead of storing the value in AdvertisementRecordType.IsConnectable as done in iOS.

janusw, 2023-05-07:
> The device should provide IsConnectable flag. This bool is available on Android in ScanResult and on iOS in CBAdvertisement.

Sounds like a reasonable enhancement.

> I would prefer this device solution instead of storing the value in AdvertisementRecordType.IsConnectable as done in iOS.

Makes sense, and should be possible. Related comment:

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/2e8615de6515fa2593ee39004a84116ec636b8e3/Source/Plugin.BLE/Shared/AdvertisementRecord.cs#L178-L181

PRs welcome ;)

midoragh, 2023-05-08:
I will create PR. I already tested that Androi provides a good value by adding it 0xAA to the advertised records.

midoragh, 2023-05-09:
Seems I do not have write access to create a branch.

midoragh, 2023-05-10:
VS2022 Windows Community
Error; The target "GetApkRecipeFile" does not exist in the project.; BLE.Client.Droid; QWERTY\dble-IsConnectable\dble-IsConnectable\Source\MvvmCross.Plugins.BLE\MvvmCross.Plugins.BLE.csproj

janusw, 2023-05-10:
> VS2022 Windows Community Error; The target "GetApkRecipeFile" does not exist in the project.; BLE.Client.Droid; QWERTY\dble-IsConnectable\dble-IsConnectable\Source\MvvmCross.Plugins.BLE\MvvmCross.Plugins.BLE.csproj

That error does not tell me much. It happens when you do what and where?

Note that there are CI builds running for every commit on the repo, which test the build with VS2022 on Windows, so we're pretty sure that the build process works fine on master. If you have trouble building master on Windows, it is most probably due to your local setup. If master works and the errors only occur on your branch, then it must be due to the changes you made.

Btw, it seems you named your fork of the repo `dble-IsConnectable` (after the feature which you want to implement). That does not really sound like a good idea to me (but is also not a real problem). Your fork can easily have the same name as the upstream repo (it can still be distinguished via your user name). Instead, the branch that you work on should be named after the feature.

midoragh, 2023-05-10:
I just compiled the fork to get the error. Found one reference regarding GetApkRecipeFile. They are talking about c++ targets from c# and relative outputpaths. Let's see if I can fix. But I'm using a fresh copy of VS 2022 with really no modifications.

midoragh, 2023-05-10:
I disabled the following in the VS22 program folder and it compiles now.

But still I'm getting this error on deployment...

janusw, 2023-05-14:
Fixed via #688.

closed 2023-05-14 by janusw