dotnet-bluetooth-le/dotnet-bluetooth-le #690: Provide Bonding properties and methods (Android)

midoragh, 2023-05-15:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce
None

## Expected behavior
Support for bonding (Android)
Device: BondState, CreateBond(), RemoveBond()
Adapter: DeviceBondStateChanged, BondedDevices

## Actual behavior
There is no support for bonding (Android)

### Crashlog
No crash

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
2.1.2

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
Android api 26

**Device:** e.g. HTC Sensation /i Phone 7 ...
Wiko

midoragh, 2023-05-15:
I would create a PR, but Ißm really not familar how it's done here.
I can not create a new fork, so I could reuse the one from IsConnectable, sync it with master and continue there.
Is this the preferred way?

janusw, 2023-05-15:
> ## Expected behavior
> 
> Support for bonding (Android) Device: BondState, CreateBond(), RemoveBond() Adapter: DeviceBondStateChanged, BondedDevices

There is a TODO note and some commented out code here:

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/48670d2c14b7bafac4b0351f85e532cff02ab91d/Source/Plugin.BLE/Android/Adapter.cs#L31-L40

Apart from that, there seems to be nothing yet.

janusw, 2023-09-21:
This is fixed via #730.

closed 2023-09-21 by janusw