dotnet-bluetooth-le/dotnet-bluetooth-le #697: Restore nuget packages on mac in Xamarin.forms project

EhMidtJyde, 2023-06-13:
So we have a Xamarin.forms project, where we try to update to latest version of bluetooth le, to prepare for MAUI later.

On windows and android everything works perfectly, I can restore packages, build and bluetooth LE works.

But on Mac when I try to restore packages in my Xamarin.forms project, it comes with error:
"error NETSDK1045: The current .NET SDK does not support targeting .NET Core 6.0.  Either target .NET Core 5.0 or lower, or use a version of the .NET SDK that supports .NET Core 6.0."

I have found a workaround, if I remove all references to Plugin.Ble, and restore then it restore everything correct, afterwards I can build and deploy.

It is easy to reproduce if you have macbook (At least M1) and visual studio for mac. By creating empty Xamarin.forms project, and add Plugin.Ble to iOS and Forms project, try it out.

I have tried to enable and disable "Build with MSBuild on mono", both with same result.

Off topic:
I actually have 2 problems in the end, because if I restore without references, and afterwards build. Then Xamarin.iOS is build as .netstandard project, and not XamarinIOS10, meaning adapter and so on is not implemented. On windows it does build as XamarinIOS10, so maybe there is more problems in the Plugin.BLE project file target platforms on mac ?
But I will create another ticket on that, when restore is fixed.

## Steps to reproduce

1. Create empty Xamarin.forms project 

2. Add Plugin.Ble into solution

3. Add reference to project into iOS and Forms project.

4. Try to restore

## Expected behavior
Restore packages without problems

## Actual behavior
Fail with:
"error NETSDK1045: The current .NET SDK does not support targeting .NET Core 6.0.  Either target .NET Core 5.0 or lower, or use a version of the .NET SDK that supports .NET Core 6.0."

### Crashlog

## Configuration

dotnet --info:
.NET SDK:
 Version:   7.0.302
 Commit:    990cf98a27

Runtime Environment:
 OS Name:     Mac OS X
 OS Version:  13.4
 OS Platform: Darwin
 RID:         osx.13-arm64
 Base Path:   /usr/local/share/dotnet/sdk/7.0.302/

Host:
  Version:      7.0.5
  Architecture: arm64
  Commit:       8042d61b17

.NET SDKs installed:
  6.0.408 [/usr/local/share/dotnet/sdk]
  7.0.302 [/usr/local/share/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.0 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 6.0.16 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 7.0.5 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.0 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.16 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.5 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]

Other architectures found:
  None

Environment variables:
  Not set

global.json file:
  Not found

Learn more:
  https://aka.ms/dotnet/info

Download .NET:
  https://aka.ms/dotnet/download

IanBUK, 2023-07-01:
This seems very similar to my issue, in https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/702

janusw, 2023-07-03:
> Fail with: "error NETSDK1045: The current .NET SDK does not support targeting .NET Core 6.0. Either target .NET Core 5.0 or lower, or use a version of the .NET SDK that supports .NET Core 6.0."
> [..]
> Visual studio for mac version: 17.5.6

This error is sort of a known problem with VS4Mac. However, I have only seen it with mixed solutions (.NET6 and classic Xamarin), not with Xamarin-only solutions.

Can you update to VS for Mac v17.6, and check if it still occurs? It may help to add
```
<BuildWithMSBuildOnMono>true</BuildWithMSBuildOnMono>
```
to your Xamarin.Forms project.

EhMidtJyde, 2023-07-03:
> This error is sort of a known problem with VS4Mac. However, I have only seen it with mixed solutions (.NET6 and classic Xamarin), not with Xamarin-only solutions.
>
> Can you update to VS for Mac [...]

I have 17.6 now, still and issue.
Have tried MS build mono without luck.

So looks like we can only wait for VS/Nuget to fix it, or go MAUI.
We hoped to update Plugin.BLE to latest before going to convert to MAUI.
Worst part of all is, everything works on windows, for both iOS and Android

janusw, 2023-07-03:
> So looks like we can only wait for VS/Nuget to fix it, or go MAUI. 

Well, I can actually not reproduce your problem. Using the Plugin.BLE nupkg (v3.0.0-beta.4) in a pure Xamarin.Forms project works well for me, also with VS for Mac.

> ## Steps to reproduce
> 
>     1. Create empty Xamarin.forms project
> 
>     2. Add Plugin.Ble into solution
> 
>     3. Add reference to project into iOS and Forms project.
> 
>     4. Try to restore

Could you be more precise about the repro?

About step 2: You don't add a package reference to a solution, but to a project. It should only be needed in the Forms base project (.NET Standard), not the iOS or Android projects.

About step 3: Reference to which project? The Plugin.BLE package should only be added to the Froms project, not the iOS one.

EhMidtJyde, 2023-07-03:
> Well, I can actually not reproduce your problem. Using the Plugin.BLE nupkg (v3.0.0-beta.4) in a pure Xamarin.Forms project works well for me, also with VS for Mac.
>
>
> Could you be more precise about [...]

2. I add your repo to solution (Not as nuget, but all source files, and add as reference and not nuget in 3.)
3. We more or less only use Plugin.BLE in native, we have combi project atm. since we'r converting to forms atm (To get easiere to MAUI later) But result is same if I add reference in iOS or only forms. Both give same error.

janusw, 2023-07-03:
> 2. I add your repo to solution (Not as nuget, but all source files, and add as reference and not nuget in 3.)

Ok, misunderstood this part. So, yeah, this will probably not work at present. Building the full Plugin.BLE solution with VS4Mac does not work either currently. Blame Microsoft. I hope they will finally fix their tooling on Mac at some point. If you find any workaround, please let us know.

EhMidtJyde, 2023-07-03:
> Ok, misunderstood this part. So, yeah, this will probably not work at present. Building the full Plugin.BLE solution with VS4Mac does not work either currently. Blame Microsoft. I hope they will [...]

Okey fair, maybe we'r able to make workaround where we only edit Plugin.BLE on windows, since we don't touch that solution much.

IanBUK, 2023-07-06:
At least it’s not just me!

IanBUK, 2023-07-23:
So...I've added a Maui sample. It's Android only at the moment and a very simple implementation. It simply lists the devices it's found, showing ID, Device ID and Advertisement data. 

You have to enable all Bluetooth and location permissions, but it works for me. I'm using macOS on an M1 MacBook Air, with VS2023 and a Samsung A8 tablet. 

It lists all the devices it can see, which is a massive step forward for me. It looks like most of my problems were related to MvvMCross, rather than PlugIn.BLE.

https://github.com/IanBUK/dotnet-bluetooth-le

That's my fork, feel free to have a play.

EhMidtJyde, 2023-07-25:
> So...I've added a Maui sample. It's Android only at the moment and a very simple implementation. It simply lists the devices it's found, showing ID, Device ID and Advertisement data.
>
> You have to [...]

Our problem is only on Xamarin, we tried creating a clean MAUI project, we don't have the issues there.
So this is only Xamarin issue.