dotnet-bluetooth-le/dotnet-bluetooth-le #698: Bluetooth scan not return with iOS 16.4+

carlh98, 2023-06-15:
Hi, in the app, I use  `await _bluetoothLE.Adapter.StartScanningForDevicesAsync();` to search for devices.

Since I updated my iPhone with iOS 16.4.1, I can't see my ble device in the discovered list.
When I test with my another iPhone with iOS 15, I can see the name of the ble device.

## Expected behavior
I should see the name of the device I want to connect

## Actual behavior
I do not see the name of the device I want to connect

## Configuration

**Version of the Plugin:** e.g. [3.0.0-beta.3](https://www.nuget.org/packages/Plugin.BLE/3.0.0-beta.3)

**Platform:** e.g. iOS 16.4.1

carlh98, 2023-06-15:
Looking online, iPhone made a Bluetooth security fix, I think it's related to the issue that I'm having.

janusw, 2023-07-03:
I have not seen any problems with device discovery on iOS 16. IIUC the problem only occurs with a specific remote device? (If yes, which kind of device is that?) Or do you have problem discovering _any_ remote device?

carlh98, 2023-07-18:
@janusw yes it's a specific device, it's a custom BLE sensor. I'll send a message to the manufacturers to see if they can do something about it.

Looks like iOS 16.4+ is causing issue to other devs
https://developer.apple.com/forums/thread/728842

carlh98, 2023-08-10:
Looks like my issue is related to https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/468

TAMHAN, 2023-10-01:
Esteemed Quorum,
please permit me to post a question related to iOS operations with the Bluetooth Plugin. Firstly, my client and I use your product successfully on both Android and iOS 15. Sadly, on iOS 16, our scans fail to return the device names.

## Steps to reproduce
I have already read up a bit on the Apple forums, and this seems to be a widely known problem. For solving it, it is said that a parameter allowing duplicate results has to be set.

Having looked for the parameter, I found this file:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/Plugin.BLE/Apple/Adapter.cs

Sadly, I am not sure how to affect the allowDuplicates flag:
```

        protected override async Task StartScanningForDevicesNativeAsync(ScanFilterOptions scanFilterOptions, bool allowDuplicatesKey, CancellationToken scanCancellationToken)
        {
#if NET6_0_OR_GREATER || MACCATALYST
            await WaitForState(CBManagerState.PoweredOn, scanCancellationToken).ConfigureAwait(false);
#else

```

Furthermore, I also saw the @janusw commit at https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/711. Sadly, neither me nor my client know how to use GitHub well, and are desperate.

How can we aid in fixing the issue at hand? We have already raised issues with both Espressif and Apple.

## Expected behavior
The iOS 16 version of the app should behave like iOS 15

## Actual behavior
No device names are returned

### Crashlog
n/a. No crash takes place

## Configuration
 
**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
3.0.0 rc1

**Platform:**
iOS 15 and Android 13 work
iOS 16 does not work

**Device:**
American iPhone SE running iOS 16