dotnet-bluetooth-le/dotnet-bluetooth-le #700: I can't find any bluetooth devices on android sdk 30 or better

nowlights, 2023-06-22:
## Expected behavior
Find bluetooth devices

## Actual behavior
No bluetooth devices found

## Configuration

**Version of the Plugin:** 3.0.0-beta.4

**Platform:** All devices sdk 30 or higher (android 11 or higher)

When I'm testing on Android 8.0, I find the devices normal, but when testing on androids 11 or higher (currently testing on android 13) I can't find anything, it doesn't give me any errors either

My permissions configuration im Manisfest.xml

`	
	<uses-sdk android:minSdkVersion="23" android:targetSdkVersion="33" />

        
	<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
	<uses-permission android:name="android:permission.BLUETOOTH_CONNECT" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />

		
	<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
	
	
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

	
	<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
	
	<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />`

volcano619, 2023-06-26:
+1 here

robotdoly, 2023-06-27:
I have the same problem.

janusw, 2023-07-03:
> **Version of the Plugin:** 3.0.0-beta.4
> 
> **Platform:** All devices sdk 30 or higher (android 11 or higher)

AFAIK there are no (major) API changes regarding Bluetooth in Android 11 / SDK level 30. However, with Android 12 / SDK level 31 (and above) there are new runtime permissions that you will need to handle, see https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

> When I'm testing on Android 8.0, I find the devices normal, but when testing on androids 11 or higher (currently testing on android 13) I can't find anything, it doesn't give me any errors either

Actually the most important question is which API level you are targeting. Can you tell us which one that is?

nowlights, 2023-07-03:
In Android 8.0 i used API 26, in Android 13.0 i used API 33

> AFAIK there are no (major) API changes regarding Bluetooth in Android 11 / SDK level 30. However, with Android 12 / SDK level 31 (and above) there are new runtime permissions that you will need to handle, see https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

But my permissions on my issue incorrect? I saw android permissions and followed plugin.BLE documentation and android permissions

janusw, 2023-07-03:
> > Actually the most important question is which API level you are targeting. Can you tell us which one that is?
> 
> In Android 8.0 i used API 26 (I can usually find it.), in Android 13.0 i used API 33 My Target API Manifest configuration is: `<uses-sdk android:minSdkVersion="23" android:targetSdkVersion="33" />`

Ok, so no API level 30 involved. Then it's understandable.

> But my permissions on my issue incorrect? I saw android permissions and followed plugin.BLE documentation and android permissions

The manifest part looks ok at first sight, but you also need to request the permissions at runtime. You can check how our sample client does it:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/d5315bf2bfdc838f1f33b02eb386a278ed016019/Source/BLE.Client/BLE.Client.Droid/Helpers/PlatformHelpers.cs#L12-L40

nowlights, 2023-07-03:
I cant find `(Android.Manifest.Permission.BluetoothScan, true),
            (Android.Manifest.Permission.BluetoothConnect, true)`

in `Android.Manisfest.Permission`, and, i cant find `Android.OS.BuildVersionCodes.S` on `PlataformHelpers`

janusw, 2023-07-03:
> I cant find `(Android.Manifest.Permission.BluetoothScan, true), (Android.Manifest.Permission.BluetoothConnect, true)`
> 
> in `Android.Manisfest.Permission`, and, i cant find `Android.OS.BuildVersionCodes.S` on `Android.OS.BuildVersionCodes`

Well, you'll need to target API level 31 at least, otherwise they won't be available.

nowlights, 2023-07-03:
is not here on ManifestFile where i define target API?

`<uses-sdk android:minSdkVersion="23" android:targetSdkVersion="33" />`

nowlights, 2023-07-03:
I changed target API level to 31, so now i have this error:

`'You need to declare using the permission: android.permission.BLUETOOTH_CONNECT in your AndroidManifest.xml'`, but i declare permission on AndroidManifest.xml... i tried to Rebuind my Android project

janusw, 2023-07-03:
There not much advice I can give here. The error message is pretty clear. Clean up everything, rebuild, check for typos ...