dotnet-bluetooth-le/dotnet-bluetooth-le #702: Building samples on M1 Mac

IanBUK, 2023-06-24:
## Steps to reproduce

1. Clone repo on M1 Mac
2. Create a fork at [https://github.com/IanBUK/dotnet-bluetooth-le.git](https://github.com/IanBUK/dotnet-bluetooth-le.git)
3. Open in Visual Studio 2022 or Rider
4. Attempt to build solution
5. Remove BLE.Client.iOS, BLE.Client.macOS, BLE.Client.UWP and Plugin.BLE.Tests to ignore errors
6. open a terminal
7. run `dotnet restore ble.sln`
  warning given:
`/Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le/Source/BLE.Client/BLE.Client.Droid/BLE.Client.Droid.csproj : warning NU1503: Skipping restore for project '/Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le/Source/BLE.Client/BLE.Client.Droid/BLE.Client.Droid.csproj'. The project file may be invalid or missing targets required for restore. [/Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le/Source/ble.sln]
`

8. Follow instructions at: https://learn.microsoft.com/en-us/dotnet/core/tools/sdk-errors/netsdk1045#visual-studio-for-mac to untick the 'Build with MSBuild on Mono' option
9. Right-click on solution in tree
10. Choose build. 
11. Open a terminal in the root of the project, inside Visual Studio Mac 2022
12. run `dotnet restore ble.sln`
13. run `dotnet build ble.sln`

## Expected behavior
Solution should build

## Actual behavior
Error given at step 10:

`Getting restore information for project /Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le/Source/BLE.Client/BLE.Client.Droid/BLE.Client.Droid.csproj
/Users/ianbrown/Library/Caches/VisualStudio/17.0/MSBuild/11428_3/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(127,5) : error NETSDK1045: The current .NET SDK does not support targeting .NET Core 6.0.  Either target .NET Core 5.0 or lower, or use a version of the .NET SDK that supports .NET Core 6.0.
/Users/ianbrown/Library/Caches/VisualStudio/17.0/MSBuild/11428_3/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(127,5) : error NETSDK1045: The current .NET SDK does not support targeting .NET Core 6.0.  Either target .NET Core 5.0 or lower, or use a version of the .NET SDK that supports .NET Core 6.0.
Unable to create package spec for project. '/Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le/Source/BLE.Client/BLE.Client.Droid/BLE.Client.Droid.csproj'`

At step 13, the following error is given:
/Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le.mine/Source/BLE.Client/BLE.Client.Droid/BLE.Client.Droid.csproj(141,3): error MSB4019:The imported project "/usr/local/share/dotnet/sdk/7.0.203/Xamarin/Android/Xamarin.Android.CSharp.targets" was not found. Confirm that the expression in the Import declaration "/usr/local/share/dotnet/sdk/7.0.203//Xamarin/Android/Xamarin.Android.CSharp.targets" is correct, and that the file exists on disk.

### Crashlog

No crash

## Configuration

**Version of the Plugin:** cloned from GitHub at commit: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/d5315bf2bfdc838f1f33b02eb386a278ed016019

**Platform:** macOS Ventura 14.1

**Device:** MacBook Air M1

IanBUK, 2023-07-01:
### Steps to reproduce

1. Clone repo on M1 Mac  from: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le
2. Open in Visual Studio 2022
3. Build solution

### Expected Behaviour
Code builds

### Actual Behaviour
Build errors:

`Getting restore information for project /Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le.3/Source/BLE.Client/BLE.Client.Droid/BLE.Client.Droid.csproj
/Users/ianbrown/Library/Caches/VisualStudio/17.0/MSBuild/5246_3/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(127,5) : error NETSDK1045: The current .NET SDK does not support targeting .NET Core 6.0.  Either target .NET Core 5.0 or lower, or use a version of the .NET SDK that supports .NET Core 6.0.
/Users/ianbrown/Library/Caches/VisualStudio/17.0/MSBuild/5246_3/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(127,5) : error NETSDK1045: The current .NET SDK does not support targeting .NET Core 6.0.  Either target .NET Core 5.0 or lower, or use a version of the .NET SDK that supports .NET Core 6.0.
Unable to create package spec for project. '/Users/ianbrown/Documents/GitHubRepos/dotnet-bluetooth-le.3/Source/BLE.Client/BLE.Client.Droid/BLE.Client.Droid.csproj'`

### Crashlog

No crash, fails to compile

### Configuration

**Version of the Plugin:** cloned from GitHub at commit: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/d5315bf2bfdc838f1f33b02eb386a278ed016019

**Platform:** macOS Ventura 14.1

**Device:** MacBook Air M1

janusw, 2023-07-03:
Visual Studio for Mac does have problems with mixed (.NET6 + Xamarin) solutions. Some of them are fixed with v17.6, it seems, but cases like ours (with a multi-targeting project that includes both .NET6 and Xamarin) still seems to be problematic.

We have a [BLE.Mac.slnf](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/BLE.Mac.slnf) which works around this issue by removing the Xamarin samples. It can be used to build the core libraries. Building the samples with VS for Mac is not supported at present, I'm afraid. You could try if it works on the command line (via `dotnet build` etc).