dotnet-bluetooth-le/dotnet-bluetooth-le #704: Cannot set a breakpoint in event handler

derek-willis, 2023-06-28:
I am using the 'DeviceDiscovered' event with a method to handle detected devices in response to 'StartScanningForDevicesAsync()'. I am unable to debug the event code because it just doesn't like having breakpoints on any line of code in the event handler. If I do not have any breakpoints then the event handler executes perfectly, soon as there is a breakpoint, 'Break Mode' occurs and displays the message "No compatible code running on the thread", pressing 'Continue' repeats this until I remove the breakpoint, then all runs well.

        _adapter.DeviceDiscovered += Adapter_DeviceDiscovered;

        public void Adapter_DeviceDiscovered(object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e)
        {
            deviceList.Add(e.Device);
            Console.WriteLine(e.Device.Id + " : " + e.Device.Name + " : " + e.Device.Rssi);
        }

## Steps to reproduce

1.   Add the code to the application like above

2.   Put a breakpoint on any of the two lines in the event handler

3.   Observe the message window in the IDE

## Expected behavior
I should be able to break on one of the breakpoints

## Actual behavior
Cannot break at all, the description above applies

### Crashlog

No crashlog

## Configuration

**Version of the Plugin:** 3.0.0-beta.4 and beta.3 (tried both)

**Platform:** Android 9.0 - API 28

Samsung Note 8

Microsoft Visual Studio Community 2022 (64-bit), Version 17.6.4 
Framework net7.0-android

janusw, 2023-07-05:
I have no trouble breakpointing a `DeviceAdvertised` handler in a Xamarin.Forms app using Visual Studio for Mac 17.6 (executing on both Android and iOS). Are you debugging on Windows or Mac?

This is most probably a VS/MAUI bug, I assume.

axa88, 2023-11-27:
@derek-willis 
Ive unfortunately had this happen too many times using xf, not using this library at all.

If looking, you will find many similar issues as this elsewhere, [here's](https://github.com/dotnet/maui/issues/16189) a recent one

But it isnt even Maui related either.
This is what prompted me to migrate to MAUI.
Heartbreaking to see its happening there as well.

Id recommend the issue be closed and help focus on relevant ones