dotnet-bluetooth-le/dotnet-bluetooth-le #710: Adapter.ConnectToKnownDeviceAsync throws A task was canceled

juniorsaraviao, 2023-07-04:
I'm using `ConnectToKnownDeviceAsync` method but it throws a Task was canceled exception

```c#

using var cts = new CancellationTokenSource(TimeSpan.FromSeconds(10));
var param = new ConnectParameters(true, true);
await adapter.ConnectToKnownDeviceAsync(deviceId, param, cts.Token);

```

## Steps to reproduce

1. Scan for bluetooth devices.

2. Connect to specific bluetooth device using the above code.

3. It throws the exception. (The cts token is not canceled)

## Expected behavior
Connect a BLE device to the app.

## Actual behavior
The BLE device is not connected to the app.

### Crashlog
```json
"exception": {
    "type": "System.Threading.Tasks.TaskCanceledException",
    "message": "A task was canceled.",
    "stackTrace": "  at Plugin.BLE.Abstractions.Utils.TaskBuilder.SafeEnqueueAndExecute[TReturn] (System.Action execute, System.Threading.CancellationToken token, System.Threading.Tasks.TaskCompletionSource`1[TResult] tcs) <0x102428090 + 0x00a0b> in <ac1ce5528dad47caabfae66238ebe508#333df53e83bd2f570ce1b9015cfbcd2a>:0 \n  at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token) <0x1024252e0 + 0x00793> in <ac1ce5528dad47caabfae66238ebe508#333df53e83bd2f570ce1b9015cfbcd2a>:0 \n  at Plugin.BLE.Abstractions.AdapterBase.ConnectToDeviceAsync (Plugin.BLE.Abstractions.Contracts.IDevice device, Plugin.BLE.Abstractions.ConnectParameters connectParameters, System.Threading.CancellationToken cancellationToken) <0x1023d44b0 + 0x00a2f> in <ac1ce5528dad47caabfae66238ebe508#333df53e83bd2f570ce1b9015cfbcd2a>:0 \n  at Plugin.BLE.iOS.Adapter.ConnectToKnownDeviceAsync (System.Guid deviceGuid, Plugin.BLE.Abstractions.ConnectParameters connectParameters, System.Threading.CancellationToken cancellationToken) <0x102d66630 + 0x0076f> in <680562e23cb3427d811efbb788ea743a#333df53e83bd2f570ce1b9015cfbcd2a>:0 \n  at Syne.Services.Bluetooth.BluetoothService.ConnectToKnownDevice (System.Guid localDeviceId) <0x101732ba0 + 0x0055b> in <4738d7c488e646249acdac48d51f4e9c#333df53e83bd2f570ce1b9015cfbcd2a>:0 ",
    "wrapperSdkName": "appcenter.xamarin"
  }

```

## Configuration

**Version of the Plugin:** 2.2.0-pre2

**Platform:** iOS 16.4.1 and above - Android v12

**Device:** iPhone 12 Pro - Samsung Galaxy A11

janusw, 2023-07-05:
> I'm using `ConnectToKnownDeviceAsync` method but it throws a Task was canceled exception
> 
> ```cs
> using var cts = new CancellationTokenSource(TimeSpan.FromSeconds(10));
> var param = new ConnectParameters(true, true);
> await adapter.ConnectToKnownDeviceAsync(deviceId, param, cts.Token);
> ```

Possibly it is simply cancelled because your timeout expires? Can you check if using a longer timespan solves the problem and lets you connect to the device?

> **Version of the Plugin:** 2.2.0-pre2

Note: This is not a very reasonable version to work with. I'd recommend to either use the latest stable version (2.1.3) or the latest beta version (3.0.0-beta.4).

jbhardm1, 2023-07-21:
We're also facing the same issue. Please tell me some work around for this issue and when it could be resolved.

exendahal, 2023-09-15:
I am facing the same issue. I am using 2.1.3 for Xamarin Forms.
Here is my use case: 
1. Scan Ble list 
2. Connect to a device (Device connects here)
3. Turn off Bluetooth
4. Turn on Bluetooth
5. Scan Ble list
6. Connect to a device (The device doesn't connect here)

Error message:
Adapter.ConnectToKnownDeviceAsync throws A task was canceled

Umer-Mahmood-889, 2023-09-23:
Same issue here, but not on all android devices.

tylerdickerson, 2023-10-12:
This same issue also affects Windows 10.0.19041.0 in the latest 3.0.0 stable release and causes the app to crash when either ConnectToKnownDeviceAsync or ConnectToDeviceAsync are called.

Stack trace:
```
   at Plugin.BLE.Abstractions.Utils.TaskBuilder.<SafeEnqueueAndExecute>d__11`1.MoveNext() in Plugin.BLE.Abstractions.Utils\TaskBuilder.cs:line 100
   at Plugin.BLE.Abstractions.Utils.TaskBuilder.<FromEvent>d__9`3.MoveNext() in Plugin.BLE.Abstractions.Utils\TaskBuilder.cs:line 30
   at Plugin.BLE.Abstractions.AdapterBase.<ConnectToDeviceAsync>d__57.MoveNext() in Plugin.BLE.Abstractions\AdapterBase.cs:line 137
   at Plugin.BLE.Abstractions.AdapterBase.<ConnectToKnownDeviceAsync>d__66.MoveNext() in Plugin.BLE.Abstractions\AdapterBase.cs:line 299
```

peter-b4, 2025-01-06:
Almost the same issue with Plugin.BLE version 3.1.0 for .NET MAUI on Android.

As described above:
- Connect to device, read/write characteristic ... all fine.
- Turn off Bluetooth
- Turn on Bluetooth
- same connection procedure as initially: ConnectToKnownDeviceAsync + RequestMtuAsync
-> the call to ConnectToKnownDeviceAsync returns a device,
-> but the subsequent call to device.RequestMtuAsync does not terminate and the "A task was canceled" problem occurs.

This happens with and without passing a CancellationToken.
When I comment out the call to RequestMtuAsync, the execution gets past the exception handling, the exception happens later, is not caught and the app crashes. 

I use
``` csharp
CancellationTokenSource cancellationTokenSource = new();
cancellationTokenSource.CancelAfter(TimeSpan.FromSeconds(10));
try {
  IDevice? device = await _bluetoothAdapter.ConnectToKnownDeviceAsync(deviceGuid, cancellationToken: cancellationTokenSource.Token);
  var result_mtu = await device.RequestMtuAsync(512);
}
catch (DeviceDiscoverException bleEx) {...}
catch (DeviceConnectionException bleEx) {...}
catch (OperationCanceledException ex) {...}
```

Stack trace
```
at Plugin.BLE.Abstractions.Utils.TaskBuilder.<SafeEnqueueAndExecute>d__11`1[[System.Int32, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext() in D:\a\dotnet-bluetooth-le\dotnet-bluetooth-le\Source\Plugin.BLE\Shared\Utils\TaskBuilder.cs:line 106
at Plugin.BLE.Abstractions.Utils.TaskBuilder.<FromEvent>d__9`3[[System.Int32, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler`1[[Plugin.BLE.Android.CallbackEventArgs.MtuRequestCallbackEventArgs, Plugin.BLE, Version=3.1.0.0, Culture=neutral, PublicKeyToken=null]], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext() in D:\a\dotnet-bluetooth-le\dotnet-bluetooth-le\Source\Plugin.BLE\Shared\Utils\TaskBuilder.cs:line 55
at Plugin.BLE.Android.Device.RequestMtuNativeAsync(Int32 requestValue) in D:\a\dotnet-bluetooth-le\dotnet-bluetooth-le\Source\Plugin.BLE\Android\Device.cs:line 397
at Plugin.BLE.Abstractions.DeviceBase`1.<RequestMtuAsync>d__33[[Android.Bluetooth.BluetoothDevice, Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065]].MoveNext() in D:\a\dotnet-bluetooth-le\dotnet-bluetooth-le\Source\Plugin.BLE\Shared\DeviceBase.cs:line 146
```