dotnet-bluetooth-le/dotnet-bluetooth-le #712: Disconnect without reason

TorstenMue, 2023-07-13:
The communication to a remote device is lost after a random number of successful communications (around 500 - 1000)

## Steps to reproduce

1. Established communication with a write characteristic and an characteristic change event (for receiving)

2. communicate 500 to 1000 times

## Expected behavior
The connection should be stable, or at least do a silent reconnect (Autoreconnect is TRUE).

## Actual behavior
The communication stops and DeviceConnectionLost without detail information is raised.

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration
** MTU Size ** 50 or 255 (no change)
** Interval ** all tested (low,normal,high) no difference
**Version of the Plugin:** 2.1.3, MvvmCross 7.1.2
**Platform:** Android 9, 11, 13
**Device:** Galaxy Note 10 lite, Hauwei JAT, Samsung galaxy J5

fruffin, 2023-07-25:
I am seeing the same issue after a few hundred calls to characteristic.WriteAsync (I am doing a device firmware update of 400kb in blocks of 232 bytes)
Same version of the libraries as in the original post, using Android 9 (Galaxy S8+) and Android 13 (Galaxy S21 Ultra)

LeeWhite187, 2023-12-06:
Did this issue get fixed?
I'm debating BLE abstractions for MAUI, and have used this in Xamarin.

BigTree92, 2024-06-02:
Caution! Important remarks / API limitations
The BLE API implementation (especially on Android) has the following limitations:

Characteristic/Descriptor Write: make sure you call characteristic.WriteAsync(...) from the main thread, failing to do so will most probably result in a GattWriteError.
Sequential calls: Always wait for the previous BLE command to finish before invoking the next. The Android API needs its calls to be serial, otherwise calls that do not wait for the previous ones will fail with some type of GattError. A more explicit example: if you call this in your view lifecycle (onAppearing etc) all these methods return void and 100% don't guarantee that any await bleCommand() called here will be truly awaited by other lifecycle methods.
Scan with services filter: On specifically Android 4.3 the scan services filter does not work (due to the underlying android implementation). For android 4.3 you will have to use a workaround and scan without a filter and then manually filter by using the advertisement data (which contains the published service GUIDs).