dotnet-bluetooth-le/dotnet-bluetooth-le #713: Index was outside the bounds of the array

vitalii-vov, 2023-07-14:
The app crashes when trying to scan devices.
This happens on the client device. Repeats only for one user. I managed to get logs from him by sending him a debug version of the program.

## Steps to reproduce

1. Start scan devices
2. Wait
3. App crushed after some seconds

## Expected behavior
App should work

## Actual behavior
App crashes

### Crashlog
```
Index was outside the bounds of the array.
  at Plugin.BLE.Android.Device.ParseScanRecord (System.Byte[] scanRecord) [0x00020] in <e84f194db96045279f39298a58a796ea>:0 
  at Plugin.BLE.Android.Device..ctor (Plugin.BLE.Android.Adapter adapter, Android.Bluetooth.BluetoothDevice nativeDevice, Android.Bluetooth.BluetoothGatt gatt, System.Int32 rssi, System.Byte[] advertisementData) [0x00017] in <e84f194db96045279f39298a58a796ea>:0 
  at Plugin.BLE.Android.Adapter+Api21BleScanCallback.OnScanResult (Android.Bluetooth.LE.ScanCallbackType callbackType, Android.Bluetooth.LE.ScanResult result) [0x00026] in <e84f194db96045279f39298a58a796ea>:0 
  at Android.Bluetooth.LE.ScanCallback.n_OnScanResult_ILandroid_bluetooth_le_ScanResult_ (System.IntPtr jnienv, System.IntPtr native__this, System.Int32 native_callbackType, System.IntPtr native_result) [0x00011] in <5437bca4f0e742d0a18eb9a3847ccdb6>:0 
  at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.1(intptr,intptr,int,intptr)
```

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** Android 13 (One UI 5.1)

**Device:** Samsung Galaxy S21 FE 5G, model: SM-G990E/DS

vitalii-vov, 2023-07-17:
The cause of the fall of the application was a wrist bluetooth watch. After the clock was turned off, the application stopped crashing. After turning on the clock, the application no longer crashes.
It turns out that an incorrectly working device just nearby can break the application.
Perhaps you should add an exception handler to the code or additionally check advertisementData for data correctness.

vitalii-vov, 2023-07-20:
I made a clone of the library. Made the following changes to the code:
```C#
namespace Plugin.BLE.Android
{
    public class Device : DeviceBase
    {
        //    some code ...
        public Device(Adapter adapter, BluetoothDevice nativeDevice, BluetoothGatt gatt, int rssi, byte[] advertisementData = null) : base(adapter)
        {
            Update(nativeDevice, gatt);
            Rssi = rssi;
            try
            {
                AdvertisementRecords = ParseScanRecord(advertisementData);
            }
            catch (Exception ex)
            {
                WriteErrorLog(nativeDevice, advertisementData);
                Trace.Message("Filed to parse advertisementData: {0}", ex.Message);
                Trace.Message("Filed to parse advertisementData: {0}", ex.StackTrace);
            }
            _gattCallback = new GattCallback(adapter, this);
        }

        private void WriteErrorLog(BluetoothDevice nativeDevice, byte[] advertisementData)
        {
            try
            {
                var downloadsFolder = Path.Combine(Environment.ExternalStorageDirectory.AbsolutePath, Environment.DirectoryDownloads);
                var path = Path.Combine(downloadsFolder, $"ParseScanRecordErrorLog_{DateTime.Now:yyyyMMddhhmmss}.log");
                var str = "";
                str += $"Name: {nativeDevice.Name}\n";
                str += $"Address: {nativeDevice.Address}\n";
                str += $"BluetoothClass: {nativeDevice.BluetoothClass}\n";
                str += $"adsData: {BitConverter.ToString(advertisementData)}\n";
                File.WriteAllText(path, str);
            }
            catch { }
        }
    }
}
```

In the logs:
```
Name: 
Address: B0:10:A0:94:72:1D
BluetoothClass: 0
adsData: 35
```

We see advertisementData is not correct.
I propose to make changes to the code, to the constructor of the Device class, so that it looks something like this:

```C#
public Device(Adapter adapter, BluetoothDevice nativeDevice, BluetoothGatt gatt, int rssi, byte[] advertisementData = null) : base(adapter)
{
    Update(nativeDevice, gatt);
    Rssi = rssi;
    try
    {
        AdvertisementRecords = ParseScanRecord(advertisementData);
    }
    catch (Exception ex)
    {
        Trace.Message("Filed to parse advertisementData: {0}", ex.Message);
        Trace.Message("Filed to parse advertisementData: {0}", ex.StackTrace);
    }
    _gattCallback = new GattCallback(adapter, this);
}
```

janusw, 2023-07-20:
> We see advertisementData is not correct. I propose to make changes to the code, to the constructor of the Device class, so that it looks something like this:

Thanks for your analysis, @vitalii-vov. I agree that we should be more tolerant against faulty advertisements. Could you push your changes to a branch and create a PR?

janusw, 2023-07-20:
I'll note that a very similar problem has been reported in #567, and I think your proposed fix would help there as well.

vitalii-vov, 2023-07-21:
@janusw 
I created a PR for this fix https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/719
Can you release a stable version of the library?

janusw, 2023-07-21:
> @janusw I created a PR for this fix #719

Awesome, thank you! I'll have a look soon ...

> Can you release a stable version of the library?

Yes, we should do that soon. This is being requested with increasing frequency ;)

vitalii-vov, 2023-07-26:
Hello @janusw
Can you please tell me when do you plan to release the nuget package?

janusw, 2023-07-26:
I'll not be able to do anything within the next week, but I'll tag a new beta release in 10 days or so.
Until then you can use the nupkg generated by the Github Actions build, if you need it urgently.

janusw, 2023-08-04:
This is fixed via #719. Closing.

closed 2023-08-04 by janusw

vitalii-vov, 2023-08-05:
@janusw  Can an issue be considered closed if the update is not published to nuget? Library users will still get issues related to this bug.
Don't you think that at the beginning it is necessary to release an update of the library in nuget?
I'm sorry if I'm wrong.

janusw, 2023-08-05:
> @janusw Can an issue be considered closed if the update is not published to nuget?

Well, there are various nuances of "published to nuget". In this project we have the following:
* stable releases on nuget.org
* beta releases on nuget.org
* automatic builds on GitHub Actions (for every PR and every push to master)

Which one you want to use depends on your preference and use case.

I consider an issue "fixed" as soon as the fix is merged to master (which implies the availability of an auto-built nupkg that you can download and use locally).

In any case, I'll tag a new beta release today (which will also be uploaded to nuget.org) ...

janusw, 2023-08-05:
The fix is available in [3.0.0-beta.5](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.5).