dotnet-bluetooth-le/dotnet-bluetooth-le #717: Adapter.DeviceConnected event is fired

smersh1307n2, 2023-07-14:
When connecting to iDevice, the Adapter.DeviceConnected event is fired. If you disconnect from iDevice, the Adapter.DeviceDisconnected event is fired. If I connect again to iDevice, the event Adapter.DeviceConnected is not fired.

## Steps to reproduce

1.Start scan devices
2.Wait
3.Get IDevice bluetoothLE.
4.Connecting to IDevice. =>fired event Adapter.DeviceConnected
5.Wait
6.Disconnecteted from IDevice =>fired event Adapter.DeviceDisconnected
7.Wait
8.Connecting to IDevice. =>not fired event Adapter.DeviceConnected

## Expected behavior
Event must fired

## Actual behavior
Event not fired

### Crashlog
not available

## Configuration

**Version of the Plugin:** 3.0.0-beta.4

**Platform:** Windows 10

**Device:** PC

janusw, 2023-08-05:
Could you please check if this problem still occurs with 3.0.0-beta.5? There were some fixes for connect/disconnect issues on Windows, see #721.

axa88, 2024-05-03:
Finding a very different issue with this event under WinUI #841