dotnet-bluetooth-le/dotnet-bluetooth-le #726: ConnectToDeviceAsync hangs when bluetooth is turned off

bogardani, 2023-08-03:
ConnectToDeviceAsync hangs when the Android phone's bluetooth is turned off right before connecting to a device

## Steps to reproduce
1. Clone https://github.com/bogardani/ConnectToDeviceAsyncBug.git
2. Put breakpoint on line ConnectToDeviceAsyncBug/MainPage.xaml.cs:37
3. Turn on device bluetooth
4. Run the application
5. Accept permission requests
6. Click button on UI
7. Turn off bluetooth when breakpoint is hit
8. Continue execution

## Expected behavior
ConnectToDeviceAsync throws exception or returns

## Actual behavior
ConnectToDeviceAsync hangs indefinitely

### Crashlog

n/a

## Configuration

**Version of the Plugin:** 3.0.0-beta.4

**Platform:** Android 13.0 (API 33)

**Device:** Samsung Galaxy A53 5G

Andy-Donegan, 2023-09-04:
@bogardani Check if Bluetooth has power prior to running your command.

You should if you followed the tutorials and examined the supplied repository have the following:

public IBluetoothLE Ble { get; set; } <-- Gives you access to the bluetooth state on/off etc.

            Ble.StateChanged += (s, e) =>
            {
                Debug.WriteLine($"The bluetooth state changed to {e.NewState}"); <-- Will let you know if state changes on to off etc.
            };

then a quick check like this before running commands will ensure you do not fall over.

if (Ble.State != BluetoothState.On) <-- if bluetoothstate is not on, you can then prompt user to switch it on.

Also remember to use Try Catch throughout your project on all the commands as described at the bottom of the project repository.

Hope this helps.

bogardani, 2023-09-22:
Checking for Ble state beforehand only shortens the window in which there is a chance we run into ConnectToDeviceAsync.
Subscribing to StateChanged does not solve the problem since ConnectToDeviceAsync already is hanging and cannot be canceled. The only thing you could do is call WaitAsync(token) on the hanging task, and let your code run forward even though that's not optimal.