dotnet-bluetooth-le/dotnet-bluetooth-le #749: iOS 16 device scan does not work

TAMHAN, 2023-10-01:
Esteemed Quorum,
please permit me to post a question related to iOS operations with the Bluetooth Plugin. Firstly, my client and I use your product successfully on both Android and iOS 15. Sadly, on iOS 16, our scans fail to return the device names.

## Steps to reproduce
I have already read up a bit on the Apple forums, and this seems to be a widely known problem. For solving it, it is said that a parameter allowing duplicate results has to be set.

Having looked for the parameter, I found this file:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/Plugin.BLE/Apple/Adapter.cs

Sadly, I am not sure how to affect the allowDuplicates flag:
```

        protected override async Task StartScanningForDevicesNativeAsync(ScanFilterOptions scanFilterOptions, bool allowDuplicatesKey, CancellationToken scanCancellationToken)
        {
#if NET6_0_OR_GREATER || MACCATALYST
            await WaitForState(CBManagerState.PoweredOn, scanCancellationToken).ConfigureAwait(false);
#else

```

Furthermore, I also saw the @janusw commit at https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/711. Sadly, neither me nor my client know how to use GitHub well, and are desperate.

How can we aid in fixing the issue at hand? We have already raised issues with both Espressif and Apple.

## Expected behavior
The iOS 16 version of the app should behave like iOS 15

## Actual behavior
No device names are returned

### Crashlog
n/a. No crash takes place

## Configuration
 
**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
3.0.0 rc1

**Platform:**
iOS 15 and Android 13 work
iOS 16 does not work

**Device:**
American iPhone SE running iOS 16

janusw, 2023-10-03:
> Firstly, my client and I use your product successfully on both Android and iOS 15. Sadly, on iOS 16, our scans fail to return the device names.

Actually I cannot reproduce this problem. I tried with both iOS 16.7 and 17.0.2 (on an iPhone and an iPad), and in both cases I see various devices that have proper names (of course not all discovered devices have names, since some of them simply do not advertise any device name).

> I have already read up a bit on the Apple forums, and this seems to be a widely known problem. For solving it, it is said that a parameter allowing duplicate results has to be set.
>
> Having looked [...]

Actually I doubt that missing device names would be related to the `allowDuplicates` option, but you can try yourself if it makes any difference. Simply call `StartScanningForDevicesAsync` with the `allowDuplicatesKey` parameter, e.g. like this:

```
await adapter.StartScanningForDevicesAsync(allowDuplicatesKey: true);
```

Can you tell us whether you are missing names for *all* discovered devices, or only for a particular one? Do you see *any* device names?

Also, in order to make sure we understand what exactly you're doing, could you please try whether you see any device names with our sample apps, e.g. [BLE.Client.iOS](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/tree/master/Source/BLE.Client/BLE.Client.iOS) or [BLE.Client.Maui](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/tree/master/Source/BLE.Client/BLE.Client.Maui)? (The behavior there might possibly be different than in your custom app ...)

janusw, 2023-10-03:
This is possibly related to #698.

janusw, 2023-10-04:
> Furthermore, I also saw the @janusw commit at #711. Sadly, neither me nor my client know how to use GitHub well, and are desperate.

You can download the nupkg generated from this PR at https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/suites/16819045568/artifacts/960576123, save it to your disk and set up a local package source in VS, so that you can use the package in your project. Could you please tell us whether that PR makes any difference regarding your device-name problem?