dotnet-bluetooth-le/dotnet-bluetooth-le #750: StartScanningForDevicesAsync while the screen is off or app is minimized on Android

yasserjarouf, 2023-10-02:
The **StartScanningForDevicesAsync** doesn't detected devices while the screen is off even if the **ScanFilterOptions** is passed with the correct Service Guid.

## Expected behavior
The adapter to discover the device even if the screen is off (and while in foreground service).

## Actual behavior
It's not detecting the device while the screen is off or the app is invisible, actually, if the app screen is on and visible, the device is detected fine with the filter, there is a loop of discovering the device over and over and a sound alert is fired accordingly.

Once the device screen is switched off, the loop still works fine and the device is discovered, but after few seconds (around 20-30) the adapter is not able to discover the device unless the screen is on and the app is on active screen.

Not sure what am I missing, any suggestions? Maybe some permissions are needed?

**Version of the Plugin:** 2.1.3

**Platform:** Android 10

**Device:** Galaxy Note 9

janusw, 2023-10-03:
> The **StartScanningForDevicesAsync** doesn't detected devices while the screen is off even if the **ScanFilterOptions** is passed with the correct Service Guid.
> ## Expected behavior
> 
> The adapter to discover the device even if the screen is off (and while in foreground service).

With a scan filter, this should actually work. One suggestion ...

> **Version of the Plugin:** 2.1.3

... Could you maybe try a newer version of the plugin, ideally the recent 3.0.0-rc.1?

And a question: 

> **Platform:** Android 10

Are you actually *targeting* Android 10, i.e. API level 29? Or just running on an Android 10 device?

If the latter, which SDK level are you targeting?

yasserjarouf, 2023-10-03:
Nothing, still same behavior.
> ... Could you maybe try a newer version of the plugin, ideally the recent 3.0.0-rc.1?

This setup I just tested with same issue `<uses-sdk android:minSdkVersion="28" android:targetSdkVersion="33" />`
> And a question:
> 
> > **Platform:** Android 10
> 
> Are you actually _targeting_ Android 10, i.e. API level 29? Or just running on an Android 10 device?
> 
> If the latter, which SDK level are you targeting?

yasserjarouf, 2023-10-03:
I have found my issue. 

It's with the App permissions. I had to set it for the **Location Access** to be **Allow all the time**!

I hope it helps whomever may come across this confusion. 😊

Just something I noticed, I had to unlock my phone after the phone reboots as I'm using a Foreground Service with a Boot Recei

closed 2023-10-03 by yasserjarouf