dotnet-bluetooth-le/dotnet-bluetooth-le #752: Incorrect AdvertisementRecords on ios

vitalii-vov, 2023-10-06:
I have a device that advertises ShortLocalName and CompleteLocalName.
My task is to read ShortLocalName.
When retrieving the device.AdvertisementRecords property on Android, the result contains fields like [Flags, ShortLocalName, CompleteLocalName] and on iOS [Flags, CompleteLocalName].

## Steps to reproduce

1. Call `var record = device.AdvertisementRecords?.FirstOrDefault(a => a.Type == AdvertisementRecordType.ShortLocalName);` on ios
2. record is null

## Expected behavior
record is not null

## Actual behavior
record is null

## Configuration
**Version of the Plugin:** 2.1.3
**Platform:** iOS 17
**Device:** iPad

janusw, 2023-10-06:
So you're missing the `ShortLocalName` records. Do you see any `CompleteLocalName`?

Might possibly be related to #749 or #698. Do you have the possibility to check iOS 15?

Also, could you try if 3.0.0-rc.1 behaves the same way as 2.1.3?

janusw, 2023-10-06:
> So you're missing the `ShortLocalName` records. Do you see any `CompleteLocalName`?

Sorry, apparently I failed to read your description properly, because it clearly states that you do see the `CompleteLocalName` records 🙈

vitalii-vov, 2023-10-06:
It looks like there are restrictions on receiving advertising data on iOS. It looks like ios wraps both keys into one. https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/advertising_data
 I'll see if there are any workarounds, if not, then I'll close the issue.

vitalii-vov, 2023-10-06:
It looks like a workaround could be to get the data in RAW format, but there doesn't seem to be that option on iOS.
Can I close the issue? If you think it necessary, you can do it yourself.

closed 2023-10-06 by vitalii-vov

bodeg, 2024-08-13:
@vitalii-vov , can you share your workaround ?
I have the same problem with iOS - a missing ManufacturerSpecificData record

vitalii-vov, 2024-08-13:
@bodeg  According to what is written [here](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/09b1b2c4dafdc26f4491caa11a99b44c4d98b461/Source/Plugin.BLE/Apple/Adapter.cs#L351), the `DataManufacturerDataKey` is checked on the ios side, and supposedly, if your device sends this data, you can read it.
In my case, however, the `ShortLocalName` key I needed could not be checked on the ios side due to `Apple SDK` restrictions, as apple hides the raw data inside its sdk. I had to give up scanning `ShortLocalName` in ios project.