dotnet-bluetooth-le/dotnet-bluetooth-le #754: Paired device not being detected by call to Adapter.GetSystemConnectedOrPairedDevices

TiisangMguni, 2023-10-12:
Hi, i am running the Maui sample (BLE.Client.Maui) on Android, I have a Bluetooth to serial device ([HC06 Serial Bluetooth Brick](https://wiki.iteadstudio.com/HC06_Serial_Bluetooth_Brick)) already paired with the phone on which i am running the sample app. The Bluetooth to serial device is connected to an Arduino device that continuously writes string data to the Arduino serial port.

The issue i am having is that the device (which shows in my paired devices under the name itead) is never returned by the call to 
`Adapter.GetSystemConnectedOrPairedDevices();`. I can however connect to the device using other android apps from the google play store and read the data that it sends continuously.

## Expected behavior
Expected a list with all paired devices

## Actual behavior
The call only returns my paired Airpods

JuLights, 2023-10-14:
[image]

same problem, **GetSystemConnectedOrPairedDevices()** doesn't return anything, but I can see it(ESP32test) in android.bluetooth.BluetoothAdapter.BondedDevices.

Auto72, 2024-01-18:
I want to summarize the behavior of the Adapter.**GetSystemConnectedOrPairedDevices**() with the BLE.Plugin ver. 3.1.0-beta.

```
var pdList = Adapter.GetSystemConnectedOrPairedDevices();
foreach(var pd in pdList) 
{
    Debug.WriteLine($"Name: {pd.Name} | Id: {pd.Id} | IsConnectable: {pd.IsConnectable} |");
}
```  

On **Android** it works form me, I get the list of the paired BLE devices.

On **Windows** doesn't work form me, what it works is the following:

```
    public override IReadOnlyList<IDevice> GetSystemConnectedOrPairedDevices(Guid[] services = null)
    {
        string pairedSelector = BluetoothLEDevice.GetDeviceSelectorFromPairingState(true);
        DeviceInformationCollection pairedDevices = DeviceInformation.FindAllAsync(pairedSelector).GetAwaiter().GetResult();
        List<IDevice> devlist = ConnectedDevices.ToList();
        List<Guid> ids = ConnectedDevices.Select(d => d.Id).ToList();
        foreach (var dev in pairedDevices)
        {
            Guid id = dev.Id.ToBleDeviceGuidFromId();
            ulong bleaddress = id.ToBleAddress();
            if (!ids.Contains(id))
            {
                var bluetoothLeDevice = BluetoothLEDevice.FromBluetoothAddressAsync(bleaddress).AsTask().Result;
                if (bluetoothLeDevice != null)
                {
                    var device = new Device(this, bluetoothLeDevice, 0, id);
                    devlist.Add(device);
                    ids.Add(id);
                    Trace.Message("GetSystemConnectedOrPairedDevices: {0}: {1}", dev.Id, dev.Name);
                }
                else
                {
                    Trace.Message("GetSystemConnectedOrPairedDevices: {0}: {1}, BluetoothLEDevice == null", dev.Id, dev.Name);
                }
            }
        }
        return devlist;
    }
``` 

We need to use **BluetoothLEDevice**, instead of the current BluetoothDevice.

On iOS I get the following error when I tray to call the method as above, at the beginning:

_Value cannot be null. (Parameter 'serviceUUIDs')_

```
    public override IReadOnlyList<IDevice> GetSystemConnectedOrPairedDevices(Guid[] services = null)
    {
        // FIX: To avoid error, insted of null, we need to uspecify new CBUUID[1];
        CBUUID[] serviceUuids = null;
        if (services != null)
        {
            serviceUuids = services.Select(guid => CBUUID.FromString(guid.ToString())).ToArray();
        }

        var nativeDevices = _centralManager.RetrieveConnectedPeripherals(serviceUuids);

        return nativeDevices.Select(d => new Device(this, d, _bleCentralManagerDelegate)).Cast<IDevice>().ToList();
    }
```

Anyway, even if I fix the problem that way, it doesn't return my paired device, that I can see listed in My Devices under the Bluetooth Settings.

I need help in iOS, in order to get my paired BLE device listed by the method .GetSystemConnectedOrPairedDevices()

vchelaru, 2024-08-28:
I am also running into this problem on iOS, and the fact that it's been months with no response is very discouraging :( 

Has anyone figured out a workaround for this? How can I tell if a device is connected if not for using GetSystem.ConnectedOrPairedDevices on iOS?