dotnet-bluetooth-le/dotnet-bluetooth-le #755: ConnectToDeviceAsync unable to connect to Bluetooth devices

StevenBase, 2023-10-13:
## Steps to reproduce

1. Turn on the Bluetooth switch on your phone
2.Connect Bluetooth devices using an APP
3.After the APP is successfully connected, turn off the Bluetooth switch on the phone
4.Turn on the Bluetooth switch of the phone again
5.Reconnect the last connected Bluetooth device using the APP

## Expected behavior
If you view 'Adapter. Connected Devices', you will find that a previously connected device already exists and cannot be successfully connected again,After waiting for a period of time, an exception 'the task is cancelled' will be reported

hs8058, 2023-10-14:
Gents,

I am re-iterating this issue on Android after an upgrade from 2.1.3 to 3. I am creating a different issue around our observations anyways but it is to confirm that the behavior from 2.1.3 to 3 about ConnecToKnownDevice also has changed entirely, which caused us a lot of headache.

IainS1986, 2023-10-22:
Are you caching the Device object after the first connect and trying to reconnect to that?

Turning Bluetooth off will clear the Android ble cache, so you too should clear/null out your Device reference on disconnect and get a new instance if the Device after a new scan.