dotnet-bluetooth-le/dotnet-bluetooth-le #756: Does the BLE.Client.Maui sample work on Windows?

melihercan, 2023-10-14:
I cannot get BLE.Client.Maui sample app running on Windows. It is stated in the WEB page that MAUI is supported on all OSes, and I assume Windows is one of them:
**MAUI (all 4 OS)		>= 3.0.0**

When I build the sample and run on "Windows Machine", I get this error popup:
_The debug target 'Windows Machine' is not compatible with the active framework 'Android', please select a valid debug target for 'Android'._

I realized that in the project file, Windows is not specified as a TargetFramework!!!
`<TargetFrameworks>net7.0-android33.0;net7.0-ios;net7.0-maccatalyst</TargetFrameworks>`

What am I missing? Is the sample runs on Windows?

## Steps to reproduce

1.Get repo v3.0.0
2.Set Samples BLE.Client.Maui as active project
3.Build
4.Set 'Windows Machine' as  target
5.Run  

## Expected behavior
The sample should run

## Actual behavior
Error popup:
_The debug target 'Windows Machine' is not compatible with the active framework 'Android', please select a valid debug target for 'Android'._

### Crashlog
No crash log, just error popup message

## Configuration

**Version of the Plugin:** 3.0.0

**Platform:** Window11

**Device:** Windows11 PC

janusw, 2023-10-20:
> When I build the sample and run on "Windows Machine", I get this error popup: _The debug target 'Windows Machine' is not compatible with the active framework 'Android', please select a valid debug [...]

You're not missing anything. The sample simply does not work on Windows yet, since it has never been tested there (see #736).

As you noticed, the TargetFramework for Windows is missing, and possibly a few other things. I will try to look into this soon. You could also try yourself to make it run on Windows. Shouldn't be all that hard, I hope.

janusw, 2023-10-20:
I made a start here: #758
Unfortunately it still fails at runtime.

janusw, 2023-10-21:
I got it to work by now. Will merge the PR soon.

melihercan, 2023-10-21:
> I got it to work by now. Will merge the PR soon.

That is fantastic. Thanks for the effort...

PR #758 (2023-10-21): Make the Maui sample client work on Windows
This adds a Windows target framework to the MAUI sample client (BLE.Client.Maui), so that it can be compiled for Windows.

It also makes the client run properly on Windows, by adding a stub implementation of the `IPlatformHelpers` interface and fixing a runtime exception.

I was not able to make the GHA build work with BLE.Client.Maui on Windows yet, but I'm at least building the client in the Mac variant of GHA now. It should hopefully be added to the Windows build soon (as soon as I understand why it fails).

This fixes #756.
closed 2023-10-21 by janusw