dotnet-bluetooth-le/dotnet-bluetooth-le #757: After upgrading from 2.1.3 to 3, ConnectToKnownDeviceAsync became a single-device connectivity function with some odd observations

hs8058, 2023-10-14:
Everything was fine with 2.1.3 on Android until we had to upgrade to 3 in order to get our UWP version of a Xamarin forms app also to build and run. 

Then a range of oddities started to appear and caused a range of issues for us in production.

Most importantly, we have a range of BLE peripherals like printers and sensors that are all initialized at startup. We use ConnectToKnownDeviceAsync "without" any CancellationTokens and everything was fine with connectivity and auto-reconnect on 2.1.3 and it still is! We run multiple independent async tasks to try to connect to devices at startup. If a device is absent, the next ones will all be found consistently and are all initialized. When the absent device becomes available, we even observe that it is getting connected without an issue. Even when we disconnect them after connection, and re-connect them, still they become connected and all is healthy.

To our amazement, after upgrading to 3.0.0, we accidentally realized that in the absence of one device, ConnectToKnownDeviceAsync in one async function affects the behavior of ConnectToKnownDeviceAsync for another async function. If a device is unavailable, ConnectToKnownDeviceAsync  waits indefeintely for it to become available as expected but until the first call succeeds or fails, the next call to ConnectToKnownDeviceAsync never gets a chance to initialize and connect to another device, irrespectively, unlike v 2.1.3.

Now, with our religious attempt to force it to work with v3 and a timed CalcellationToken that we never had to use, this time we had to re-invent the wheel of running a re-try timer to try to connect to the absent devices so that upon the failure of the first one, then we have a chance to connect to the next one.

Result: it never worked. When the first device's ConnectToKnownDeviceAsync  fails on the timed CalcellationToken, then it affects the connectivity of the 2nd device that is present as if, whenever we do, the plugin at v3 can only work with one and only one device. When the first device is available and the 2nd one is unavailable, we can connect to the first one, but then when the retry timer for the 2nd device makes an attempt with ConnectToKnownDeviceAsync, then the first successful connection gets affected with dropouts and failures.

Let us be clear here: We know our onions very well and we're not fooling around and we really are time poor and we want to help. Hope you understand the mere fact of creating this ticket is a huge undertaking for us. So simply follow the instructions to recreate the situation instead of saying you can;t recreate the issue.

## Steps to reproduce

1. Use v 2.1.3 on Android, in a setup where you have more than 1 BLE device and you are initializing them all at one stage using ConnectToKnownDeviceAsync using multiple independent async functions, connect to them and observe healthy connectivity with read and write operations 

2. upgrade to v3

3. Make the first device in the queue unavailable and observe the behavior of ConnectToKnownDeviceAsync for the second one and so forth. Confirm that the 2nd and 3rd etc... calls to ConnectToKnownDeviceAsync do not succeed and remain dependent on the first call, no matter the setting.

## Expected behavior
Just like v 2.1.3, ConnectToKnownDeviceAsync must remain indefinitely waiting for the device to become available but when called on multiple async functions, it should make an attempt to connect to other devices irrespective of the previous outcomes. In other words, we should not have a single-device model for connections.

## Actual behavior
If the first call to ConnectToKnownDeviceAsync becomes stuck on an unavailable device, one can never connect to another device no matter the setting. If a timed CancellationToken is passed, the call to the first ConnectToKnownDeviceAsync does raise and exception, but after that 

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 3.0.0
**Platform:** e.g. Android 13 
**Device:** e.g. Samsung G525F.

janusw, 2023-10-16:
> Everything was fine with 2.1.3 on Android until we had to upgrade to 3 in order to get our UWP version of a Xamarin forms app also to build and run.

So, you see a difference in behavior between 2.1.3 and 3.0.0.

Inf fact there were a number of betas and pre-release versions (3.0.0-beta and 2.2.0-pre). Can you tell us which of those are still working as expected and at which point the problems start to appear? This will help to identify when and how your use case broke.

IainS1986, 2023-10-20:
> Let us be clear here: We know our onions very well and we're not fooling around and we really are time poor and we want to help. Hope you understand the mere fact of creating this ticket is a huge [...]

How **not** to post an issue to an open source, essentially volunteer maintained, free plugin you're using.

IainS1986, 2023-10-22:
Just to add some (I imagine unwanted!) Noise to this, recently we've seen some issue but on iOS in 3.0 after upgraded from 2.x

Annoyingly, it's not very reproducible, I'm just going of reports from some of our testers in the field and they were reporting devices failing to connect.

We only maintain a single device being connected, on iOS we use ConnectToKnownDevice. It seems like things work fine...until they don't :( 

A reboot of the app 'fixes' it.

Next week I'm going to try and Investigate a bit, I'm just hampered with the fact I can never reproduce it.

My hunch is it's around the 'DiscoveredDevices' cache that now exists. I wonder if native IDevice objects in there can become 'unconnectable' in some way? We never cache IDevice objects ourselves once a disconnect happens (tbh I didn't think it was safe to do that on Android at least - no idea with iOS? But Android documentation is so awful I could have just assumed it).

Either way though, might be nice to have a connection parameter to tell it to clear out any existing devices on the Discovered Devices before trying a new connection? 

I might try the 3.x beta from before that cache was introduced, but might take a few days before I get word on if that 'fixed' things or not.

IainS1986, 2023-10-23:
Actually, as one possible work around I'm going to try just calling the `AdapterBase::ConnectToKnownDeviceNativeAsync` to bypass this DiscoveredDevicesRegistry lookup, and see if that resolves the issue when using latest 3.X

janusw, 2023-10-27:
> My hunch is it's around the 'DiscoveredDevices' cache that now gets queried and used instead of creating a device from the guid.

So, the suspicion here is that the problem was introduced by #735 (by @TravisRo). If this is the case, then 3.0.0-beta.5 should work fine, while 3.0.0-beta.6 should show the misbehavior. Can somone confirm that?

IainS1986, 2023-10-27:
I can't confirm that at the moment.

What I can say, we had a few people out at a tradeshow last week on a build of the app on `3.0.0rc1`

Tradeshow was around 4 days.

First 3 days, they all experienced failures to connect every day.

I then did them a quick build that just rolled back to `2.1.3` and the last, and further testing, and they never experienced it again.

This week, we had another tradeshow, albeit smaller in scope but still 4 days.

This time I did a build with `3.0.0rc1` again, but instead of called `ConnectToKnowDeviceAsync`, I bypassed it and instead called the inner `ConnectToKnownNativeDeviceAsync` (which I believe is pretty much just the old function)

At no point over the 4 days of this week's tradeshow did they experience any issues.

It's still early doors for me to confirm but it's seeming like that's the culprit for me at least...

hs8058, 2023-10-28:
> How **not** to post an issue to an open source, essentially volunteer maintained, free plugin you're using.

Wrong answer. How not to post an issue to open source is simply not to do it at all and instead silently switch to an alternative without caring enough to actually mention it to help others. No one is prosecuting you for anything but different people have different workloads and everyone is doing what they can do. If we had enough room, we would have tried to look into the causes ourselves.

skha83, 2023-11-01:
Hi, I have experienced the same problems with connecting to two Bluetooth devices while running Android. I tried to revert to every beta release of the 3.0 version, but they all had the same problem. Then I investigated the changes made from 2.1.3 to 3.0.0. 

At some point a new shared TaskBuilder in the Plugin.BLE.Abstractions.Utils namespace was introduced which purpose seems to be to ensure only one command is executed at a time. This class is also used for the ConnectToKnownDeviceAsync method in the shared implementation here: 
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b2c3893ea65c3003addd796d128f9601759114fc/Source/Plugin.BLE/Shared/AdapterBase.cs#L110  

The problem is that the underlying ConnectToDeviceNativeAsync method on Android never returns if no successful connection to a device is made. It just sits and wait until a device become present. This means that the semaphore on the following line will never be released if a device never shows up: 
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b2c3893ea65c3003addd796d128f9601759114fc/Source/Plugin.BLE/Shared/Utils/TaskBuilder.cs#L69
 
The ConnectToKnownDeviceAsync for the second device will then never be allowed to execute. I have verified that this is indeed the problem by disabling the queuing of commands by setting the following static properties early in the app flow:

`BleImplementation.ShouldQueueOnMainThread = false;`
`Plugin.BLE.Abstractions.Utils.TaskBuilder.MainThreadInvoker = null;`

The reason why there is no problems on iOS is that the ConnectToDeviceNativeAsync method on iOS is not blocking even if no device is found, so the semaphore gets released for the second connect call to proceed.

I have not looked into what the proper solution for this problem is, I just wanted to share my findings, so people are not hunting ghosts looking at the recent PR’s which doesn’t seem to be the cause of the problem, as it seems to be introduced between version 2.1.3 and 3.0.0 beta1. I hope this information can help solve the problem.

IainS1986, 2023-11-01:
Ah that *is* interesting, and I think points to myself having a different issue with regards to iOS devices failing to connect. So @janusw you might be able to ignore my stuff as 'noise' for now (although as an update, since calling the base `ConnectToKnownNativeDeviceAsync` we haven't run into any issues).

With regards to what you've found skha83 - we never have multiple devices connected, but we also run our own Queue system anyway.

I'm guessing disabling the internal queue is 'safe' in terms of it not working any different from before. So maybe the solution for now is to make the Queue 'off' by default so nothing internally changes.

It *is* advisable (on Android especially) to have a Queue system, but a lot of people might have already solved that problem themselves in their own solutions and are happy to continue as before?

janusw, 2023-11-02:
> Ah that _is_ interesting, and I think points to myself having a different issue with regards to iOS devices failing to connect.

So yeah, probably there are two or even three separate problems reported in this thread. We need to pinpoint more precisely the source(s) of the changed behavior that you guys are seeing, in order to tell them apart.

@IainS1986, since your problem affects iOS, and seems to be different from the Android problems here, maybe it would be best if you open a new issue for it?

janusw, 2023-11-02:
> At some point a new shared TaskBuilder in the Plugin.BLE.Abstractions.Utils namespace was introduced which purpose seems to be to ensure only one command is executed at a time. This class is also used for the ConnectToKnownDeviceAsync method 

Thanks for your investigation, @skha83. The TaskBuilder was modified between 2.2.0-beta.1 and 2.2.0-beta.2 in the commits 0447b9feffe730034c93aac925f69768709f1d69 and 4fd481f4462a25b96492dfca799c4f3bf6603d02, which also introduced the property `ShouldQueueOnMainThread`.

If this is the source of your problem, then 2.2.0-beta.1 should work well for you while 2.2.0-beta.2 should not. Can you confirm that?

> I have verified that this is indeed the problem by disabling the queuing of commands by setting the following static properties early in the app flow:
> 
> `BleImplementation.ShouldQueueOnMainThread = false;` `Plugin.BLE.Abstractions.Utils.TaskBuilder.MainThreadInvoker = null;`

@hs8058 Could you check if this workaround also fixes your problem?

skha83, 2023-11-03:
Hi @janusw I have tested both version 2.2.0-pre1 and 2.2.0-pre2 and it is indeed the changes between these two versions which break the functionality. 2.2.0-pre1 works 2.2.0-pre2 doesn't. The change in itself is a good change which ensure people would experience less problems, because of the synchronization to main thread and sequential behavior of Bluetooth commands. As far as I'm aware it is only the ConnectToKnownDevice method on Android which can potential never return if no device is present. Because of this it needs special handling regarding the sequential behavior, as it can't be allowed to block all other commands. All the other Bluetooth commands would benefit from the changes, which is why my temporary fix is not really a good general solution either, but is needed for now, if your app connects to more than one device. As our code is now being converted from Xamarin to MAUI and from the old 2.1.3 version of the package, we already took care of the sequential behavior in our existing code, which is why I can use my fix, but it might not be the case for everybody.

janusw, 2023-11-07:
> Hi @janusw I have tested both version 2.2.0-pre1 and 2.2.0-pre2 and it is indeed the changes between these two versions which break the functionality. 2.2.0-pre1 works 2.2.0-pre2 doesn't.

Thanks for the confirmation.

> The change in itself is a good change which ensure people would experience less problems, because of the synchronization to main thread and sequential behavior of Bluetooth commands. As far as I'm [...]

If your analysis of the problem is correct, then the best solution would be to only disable the `MainThreadInvolker` in `ConnectToDeviceAsync`, right?

skha83, 2023-11-08:
> If your analysis of the problem is correct, then the best solution would be to only disable the `MainThreadInvolker` in `ConnectToDeviceAsync`, right?
>
Yes, and it is only necessary to do it for Android. The iOS connect method is not blocking. Windows I'm not sure as I only develop on mobile platforms. If it is simpler to implement, disabling it for this method on all platforms might be the easiest solution.

janusw, 2023-11-08:
> Yes, and it is only necessary to do it for Android. The iOS connect method is not blocking. Windows I'm not sure as I only develop on mobile platforms. If it is simpler to implement, disabling it [...]

Yes, I think it is. This whole game will only affect Android anyway (on all other platforms the `MainThreadInvolker` is not used at all). I can try to prepare a simple PR for that. Would be great if you could test it, @skha83.

skha83, 2023-11-08:
Sure I can test it if you come up with a solution. Our app always has two Bluetooth devices paired, so it is a scenario we use every day, so it is easy for me to test @janusw

janusw, 2023-11-08:
> Sure I can test it if you come up with a solution. Our app always has two Bluetooth devices paired, so it is a scenario we use every day, so it is easy for me to test @janusw

Awesome :+1:

I'm not sure what is the most elegant way to fix this, but I think I can hack up a working solution tonight. 😆

janusw, 2023-11-08:
> I'm not sure what is the most elegant way to fix this, but I think I can hack up a working solution tonight. 😆

see #774

skha83, 2023-11-09:
@janusw the generated Nuget package from that PR works perfectly 👍

janusw, 2023-11-09:
> @janusw the generated Nuget package from that PR works perfectly 👍

Great to hear that. @hs8058 Could you try this as well? (I assume your problem is the same, but I'm not 100% sure.)

TravisRo, 2023-11-09:
> Yes, and it is only necessary to do it for Android. The iOS connect method is not blocking. Windows I'm not sure as I only develop on mobile platforms. If it is simpler to implement, disabling it [...]

I know I'm late to the party but Windows doesn't need for anything to be invoked on the main thread.

janusw, 2023-11-12:
> I know I'm late to the party but Windows doesn't need for anything to be invoked on the main thread.

Correct. As mentioned above, the `MainThreadInvolker` is only used on Android (it is not needed on other platforms).

closed 2023-11-12 by janusw

RuT81, 2024-01-19:
I will be very happy if this can be fixed as soon as possible. I spent two days solving the same problem with multiple devices barcode scanner and printer. The barcode scanner usually goes to sleep to last as long as possible.
Disabling MainThreadInvoker helped resolve the situation.

IainS1986, 2024-01-19:
I believe its fixed in the `3.1.0-beta.1` build so you can try that pre-release and see if it works for you

RuT81, 2024-01-19:
Connection goes ok with 3.1.0-beta.
I'm probably making a mistake somewhere. 
I calling adapter.DisconnectDeviceAsync(device) after losing connection with the device.
Is there anything else what I need to call for clean up a old connection?

Because in Logcat there is an extra line like this for every next reconnection same device
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
which does not look good.

It's look like something doesn't clean up when the connection is lost. And after 6-8 reconnect it freeze completely.
Thanks for your help.

[DeviceBLEComm] DeviceConnected Scanner 830-GATT
[0:] [18:03:57 WRN] DeviceConnected Scanner 830-GATT
[0:] [18:03:57 INF] 18:03:57 - Scanner setup notification
[BluetoothGatt] discoverServices() - device: XX:XX:XX:XX:19:7A
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onSearchComplete() = Device=XX:XX:XX:XX:19:7A Status=0
[BluetoothGatt] setCharacteristicNotification() - uuid: 0000fff1-0000-1000-8000-00xxxxxxxxfb enable: true
[0:] [18:03:58 INF] 18:03:58 - Scanner waiting for notification
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=40 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=40 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=40 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=40 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=40 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=40 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=XX:XX:XX:XX:19:7A interval=40 latency=0 timeout=500 status=0
[0:] [18:03:58 INF] 18:03:58 - Scanner connecting

M0n7y5, 2024-01-23:
IDevice has Dispose method but not sure if that helps you.

RuT81, 2024-01-29:
I found where was my bug. I was connecting to device with autoConnect = true param. 
After lost a connection I was call ConnectToKnownDeviceAsync + StartUpdatesAsync and that added next registration on notification under the Gatt

PR #774 (2023-11-12): disable MainThreadInvoker in Adapter.ConnectToDeviceAsync
This is an attempt to fix #757. @skha83, could you please try this and tell me if it fixes the problem for you?